robotology/community #76: How to change joint names in yarpmotorgui

hu-yue, 2015-11-13:
Hi,
How can one change the name of the joints in the yarpmotorgui? I mean the ones in parenthesis after the joint number, like ```JOINT 0 (torso_roll)```.
Because in our robot (iCubHeidelberg01) the joints in the gui and the real joints do not match.
In the gui the sequence is ```roll - pitch - yaw``` but on the real robot is is ```yaw - roll - pitch```, and it makes the motorgui quite misleading everytime.
Thanks!

pattacini, 2015-11-13:
@hu-yue I think we already reported this incongruities in https://github.com/robotology/icub-main/issues/229.

hu-yue, 2015-11-13:
@pattacini yes thanks!

barbalberto, 2015-11-16:
yarpMotorGui gets the axis names from the robotInterface.
The name of the axis is the harware/mechanical/xml file in the  <param name="AxisName"> tag.
This name will also be used for ROS topic as well

hu-yue, 2015-11-16:
@barbalberto Thanks. 
I found the parameter in ```mechanicals``` instead of ```motorControl```.
However this arises a further question about the influence of the parameter. I mean, does this change affects the software of the robot?

pattacini, 2015-11-16:
Uhm, in `iCubGenova01` they're in `motorControl` ... it seems there are discrepancies then.

barbalberto, 2015-11-16:
Probably there is some difference between can-based and eth-based robots.
This difference however does not affect the robot behaviour because those two files will be 'merged' together by the code before calling the actual device.

pattacini, 2015-11-16:
It seems @barbalberto guessed well: `iCubGenova04` (ETH robot) has also those parameters under `mechanicals`.

randaz81, 2015-11-16:
@yue please do not manually edit those files. They will be fixed soon. (robotology/icub-main#229)

hu-yue, 2015-11-16:
@randaz81 I will wait for the fix. Thanks!

pattacini, 2015-11-16:
Closing this issue again then.

closed 2020-12-08 by pattacini