robotology/community #77: Commits by different users on a shared machine and repository

gsaponaro, 2015-11-18:
Hi, we would like to be able to make git commits from a machine that is shared among many users in our lab (iCub laptop). Each user has a personal GitHub account. For example, today I'd like to make a commit "signed by me", tomorrow a colleague would make another by him, etc., and all of this would happen on the single clone of the repository that exists on the machine.

We thought about changing the global file ```~/.gitconfig``` before every commit, but it's not the most practical solution. Is there a better way?

In the icub-main history, I noticed that sometimes the phrase "committed with" appears, like shown below. Perhaps that's what we'd like to achieve.

[image]

Tobias-Fischer, 2015-11-18:
Hi,

for example https://stackoverflow.com/questions/18750808/difference-between-author-and-committer-in-git and https://git-scm.com/docs/git-commit (scroll to --author).

Best, Tobias

drdanz, 2015-11-18:
If you are using different repositories you can also set the name and email in just for a specific repository, but it's not a very clean solution, and I don't know if it is the case here.

You can use `git commit --author "John Doe <john.doe@example.com>"` to change the author when you make a commit.

Another option is to use the environment variables:

```
GIT_AUTHOR_NAME
GIT_AUTHOR_EMAIL
GIT_COMMITTER_NAME
GIT_COMMITTER_EMAIL
```

"Committer" is the "iCub User" in the picture you added.

gsaponaro, 2015-11-18:
Thanks a lot. I applied the suggestions and also created Bash aliases to avoid typing the ```--author``` option every time:

```
GIT_COMMITTER_NAME="iCubLisboa team"
GIT_COMMITTER_EMAIL="username@users.noreply.github.com"
alias git-commit-as-giovanni="git commit --author=\"Giovanni Saponaro <gsaponaro@isr.ist.utl.pt>\""
alias git-commit-as-anotherUser=...
```

pattacini, 2015-11-18:
@drdanz do you know what will happen for a commit where `email` and `author` options do not match each other?

pattacini, 2015-11-18:
Uhm ... it sees that somehow the information conveyed by the `email` is quite relevant for GitHub:

https://help.github.com/articles/why-are-my-commits-linked-to-the-wrong-user

pattacini, 2015-11-18:
I've noticed now that the alias does contain the `email` too; hence there won't be any mismatch.

drdanz, 2015-11-18:
@pattacini i can confirm that github uses the author *email* to associate a commit to one user, or the author *name* if the email does not correspond to any user.
At a "git" level you will see both the *name* and the *email* in the log.

drdanz, 2015-11-18:
Closing this, @gsaponaro please reopen the issue if you need more information.

gsaponaro, 2015-11-18:
Perfect, thanks.

closed 2020-12-08 by pattacini