robotology/community #78: Kinematic chain Head- Right hand - CAD picture

vicentepedro, 2015-11-24:
Hi all!

I need the kinematic structure of iCub from the head to the right hand in order to show how the joints are placed in the robot.

I have found this picture of the head:
[image]

and this one for the right hand (with the reference frame attached)
[image]

I was wondering if there is a picture with the rest of the right arm, from the neck to the wrist.

Thanks in advance,

Pedro Vicente

pattacini, 2015-11-24:
Perhaps @super-ste and/or @robotology/icub-support-team can help you out.

super-ste, 2015-11-24:
Hi,
did you try having a look here: http://wiki.icub.org/wiki/ICubForwardKinematics

Hopefully you will find the answer there (according to the version of the robot you consider).

vicentepedro, 2015-11-24:
Yes.
I found there the two images shown in the previous post. The head and the right hand.

Unfortunately, I didn't found anything for the rest of the arm. Just some plots in matlab, not the CAD design.

pattacini, 2015-11-26:
@vicentepedro the CAD drawings are stored in our `SVN` repository here: http://sourceforge.net/p/robotcub/code/HEAD/tree/trunk/iCubHardware/mechanics/.

I think you could open the right drawing with applications like **Creo**.

@super-ste can correct me if I'm wrong, and possibly point you at the proper file.

super-ste, 2015-11-27:
While what @pattacini says is correct, you can also have a look at the instructions here on how to download the CAD files:
http://wiki.icub.org/wiki/RobotCub

Or, if you just need a screenshot or anything like that, can you elaborate more on what you actually need? Maybe it's just lacking and we can add it to the wiki...

vicentepedro, 2015-11-27:
Thanks @pattacini and @super-ste. 

I will try to install **Creo** or **ProE** and open the CAD files. I will let you know if it worked well.

Actually, the CAD pictures of the arms are missing on the wiki, maybe it's a good idea to add it.

super-ste, 2015-11-27:
Another suggestion, if you just need to visualize the robot, you can install the [CREO View Express](http://www.ptc.com/product/creo/visualization/view-mcad/extension/express)
(much lighter than CREO Parametric and for free).

The file you want to look for and open could be (after checking out the whole repository) is:

trunk/mechanics/public/icub1_icub2/icub.asm
or
trunk/mechanics/public/icub1_icub2/icub2.asm

or use any combination of the pics below

[image]
[image]
[image]

vicentepedro, 2015-11-27:
Thanks, @super-ste.

The pics seem enough for my purpose. ;)

closed 2020-12-15 by pattacini