robotology/community #79: Add yarpidl_rosmsg source files to my code or to a yarp library

plinioMoreno, 2015-11-25:
I want to use the classes defined in [rosmsg source](https://github.com/robotology/yarp/tree/master/src/idls/rosmsg/src) in order to fill the type of a message.

Using cmake and the variables defined by YARP cmake files, how should I add those files to my code?

It would be possible to add those files to a YARP library?

cc @drdanz

Tobias-Fischer, 2015-11-25:
Hi,
did you check out http://www.yarp.it/rostypes_tutorial_portable.html?

Best, Tobias

plinioMoreno, 2015-11-25:
@Tobias-Fischer, the tutorial does not fill type anywhere. In the tutorial they are executing the yarpidl_rosmsg externally. Instead of doing that, I want to write a code that given a ros message name, it is able to fill the type [void Port::promiseType 	( const Type &  typ) ](http://www.yarp.it/classyarp_1_1os_1_1Port.html#a188b9971b46203b10b2ed9e6b58f886f) in my code.

There is a class that deals with that, which is located in the [rosmsg source](https://github.com/robotology/yarp/tree/master/src/idls/rosmsg/src). And I would like to use it in my code.

drdanz, 2015-11-25:
@plinioMoreno If I understand correctly your issue, I *think* that you can do something like ``port.promiseType(Type::byNameOnWire("your_package/your_message"));``

barbalberto, 2015-11-26:
I'm not sure to understand your issue, can you be more precise on what you are trying to achieve?
Do you want to have a yarp port able to communicate with a ros topic using a specific ros message, and do everything from code?

plinioMoreno, 2015-11-26:
@barbalberto, yes, I do. @drdanz, your guess was correct and works properly.

The reason for writing code like this is to get rid of issues on the [yarp-bottle-generator](https://github.com/vislab-tecnico-lisboa/yarp-bottle-generator), that we use to generate bridging code between ROS and YARP.

Thanks

barbalberto, 2015-11-26:
So the best way to do it is to use the Publisher, which is a port already customized for usage with ROS, like the controlBoardWrapper is doing.

```    yarp::os::Publisher<my_idl_generated_message>         rosPublisherPort;           // Dedicated ROS topic publisher```

closed 2020-12-15 by pattacini