robotology/community #80: Client device for `ServerInertial`

traversaro, 2015-12-03:
The yarp devices permit to access a remote device on a computer across the network using a pair of devices : the `wrapper` is attached to the remote `wrapped` device and is used to expose the device functionality over a set of yarp ports , while the `client` is used to connect to the wrapper and provide the same C++ interface of the original device.

We have instantiations of this pattern for: 
 * ControlBoards devices : 
    * `wrapped` : [CanBusMotionControl](http://wiki.icub.org/brain/classyarp_1_1dev_1_1CanBusMotionControl.html) , [embObjMotionControl](http://wiki.icub.org/brain/classyarp_1_1dev_1_1embObjMotionControl.html)
    * `wrapper` :  [ControlBoardWrapper](http://www.yarp.it/classyarp_1_1dev_1_1ControlBoardWrapper.html)
    * `client` : [RemoteControlBoard](http://www.yarp.it/classyarp_1_1dev_1_1RemoteControlBoard.html)
 * Analog (i.e. sensor publishing a vector of floating point numbers): 
   * `wrapped` : [CanBusAnalogSensor](http://wiki.icub.org/brain/classCanBusAnalogSensor.html) , [embObjAnalogSensor](http://wiki.icub.org/iCub_documentation/classyarp_1_1dev_1_1embObjAnalogSensor.html)
* `wrapper` :  [AnalogWrapper](http://www.yarp.it/classyarp_1_1dev_1_1AnalogWrapper.html)
  * `client` : [AnalogSensorClient](http://www.yarp.it/classyarp_1_1dev_1_1AnalogSensorClient.html)

* Inertial Measurement Units-like devices (implementing `IGenericSensor` interface) : 
  * `wrapped` : [XSensMTx](http://wiki.icub.org/brain/classyarp_1_1dev_1_1XSensMTx.html)
  * `wrapper` : [ServerInertial](http://www.yarp.it/classyarp_1_1dev_1_1ServerInertial.html)
  * `client` : ? 

My question is: does a `client` that connects to ports opened by a `ServerInertial` ? 
@barbalberto

barbalberto, 2015-12-04:
Yes, **it is missing since ever.** Right now the **AnalogSensorClient can be used** for this purpose.
When I've free time, I like to fix this type of dishomogeneity and this one is in the list since long time.
IMHO for this device it would be like also to define a 'structured' type instead of simply an array of data, just to easily distinguish between acceleration, euler data and so on.

pattacini, 2015-12-07:
@barbalberto if you want to take on this development I think it'd be worth tracking it with proper tickets on the corresponding repositories.
Closing.

closed 2020-12-15 by pattacini