robotology/community #82: Wheezy support in YARP/ICUB software

traversaro, 2015-12-12:
According to the documentation on the OS supported by YARP/ICUB software [1] the Debian Wheezy distribution should not be supported anymore. 

Anyway there are several iCub that still mount a Debian Wheezy as the operating system for the PC104 / support cluster . Is this a temporary situation (an update is planned) or a long term one? 

I am asking mostly to make sure that the CoDyCo/Koroibot software remains compatible with all the OS supported by YARP/ICUB (at the moment for example we are not testing in CI our software for compatibility with CMake 2.8.11, and some bugs already emerged for this reason https://github.com/robotology/idyntree/issues/111 ). 

@drdanz @lornat75 

[1] : http://wiki.icub.org/wiki/YARP_Supported_Distributions

Tobias-Fischer, 2015-12-14:
A related question: Our iCubLondon01 is also still using Wheezy. Is it safe to update to Jessie as written in the official documentation (https://www.debian.org/releases/stable/i386/release-notes/ch-upgrading.en.html)? I didn't find any PC104 customized ISO images of Jessie ..

Best, Tobi

gsaponaro, 2015-12-14:
On iCubLisboa01 we are also running "Debian GNU/Linux 7.2 (wheezy) + iCub customizations". When we eventually upgrade, I wonder if we should go via Debian upgrade on the existing OS, or rather install & customize a new iCub pc104 image.

pattacini, 2015-12-14:
@mbrunettini might be also interested.

barbalberto, 2015-12-14:
For pc104 using the standard OS upgrade is not possible.

Tobias-Fischer, 2015-12-14:
Hmmm .. this is a big issue then, seeing that Wheezy is not officially supported anymore, and yet there is no way to upgrade PC104 to a newer version. As a matter of fact, updating the `icub-common` package on PC104 isn't possible right now, as it complains about some `QT` dependencies which cannot be resolved.

plinioMoreno, 2015-12-14:
Because of our robot Vizzy (which also has a cfw002) I also asked 
@mbrunettini about newer kernel versions (Debian 8/Ubuntu 14.04), but he 
told me that the cfw002 kernel modules are not available yet for recent 
kernel versions.

lornat75, 2015-12-21:
We have to keep supporting new debians for the pc104. I think there are problems building debian live images, but we will investigate them. 

The kernel module for the cfw002 should be compilable on new kernels, or at least I am not aware of problems (please let me know otherwise).

In the meanwhile you can disable qt on pc104 I think qt guis are not needed there.

pattacini, 2016-02-12:
Can we close this issue?

Tobias-Fischer, 2016-02-12:
I think there should be an issue somewhere around regarding the compatibility of PC104 and Debian Jessie :+1:

pattacini, 2016-02-12:
I've made a quick search without success.

gsaponaro, 2016-02-12:
https://github.com/robotology/icub-support/issues/119 ?

pattacini, 2016-02-12:
Thanks @gsaponaro ! I actually forgot to check that repo too.
Closing then.

closed 2020-12-15 by pattacini