robotology/community #84: Moving issues among repositories

pattacini, 2016-01-08:
Does anybody know an handy way to move an issue from one repository to another?

I bet @drdanz has the solution :smirk:.

drdanz, 2016-01-08:
As far as I know, it is not possible to move issues from one repository to another one. What you can do is "import" the issue in another repository as a new issue and close the other one, see for example robotology/icub-main#271 that I just moved on @randaz81's request to robotology/icub-support#149

The imported comments are from the person who performs the move, but the original commenter is shown inside the comment itself.

In order to do so, there is this tool:

  https://github.com/IQAndreas/github-issues-import

The documentation and the help are quite easy to understand, anyway if you are too lazy to read it, just edit the config.ini and run

```
  python3 gh-issues-import.py --issues <issue_number>
```
and then manually close the original issue.

pattacini, 2016-02-24:
Lately, GitHub has made available the new button [**`Move issue`**] :exclamation:

pattacini, 2016-02-24:
/cc @drdanz

drdanz, 2016-02-24:
Cool, it's a very good addition, even though it does not allow to move issues in batch

pattacini, 2016-02-29:
Well, actually this feature comes from `Zenhub`, not `Github`... :smirk: 
... but still it's very useful!
/cc @drdanz

drdanz, 2016-02-29:
Right! https://github.com/ZenHubIO/support/issues/72#issuecomment-185448363

closed 2020-12-08 by pattacini