robotology/community #85: Add new objects' model to the iCub simulator

ZHUQINGPENG, 2016-01-13:
Hi, guys,

I want to load some objects with textures into the iCub simulator. I downloaded some .x files but it doesn't work. Do you have some objects' model (like table,tree, car, house...) in .x file or can you provide some website that I can download.

Thank you very much

pattacini, 2016-01-22:
Does anyone from @robotology/all-robotology-developers community has hints on this?

tanismar, 2016-02-29:
Hi, 
Sorry for the late response, I hope the answer is still useful. I have a bunch of .x models on the following link: 
https://drive.google.com/folderview?id=0Bykt6OYDAgv_V3N1T2VMRlM5WHM&usp=sharing

They are basically tool models, but you can use them for trying if loading them on the simulator works for you (it worked for me). 

Actually, the way I got these models was using a 3D model application and export them in .x. In my case, I used SketchUp (free online) with the 3rd party Zbyl's exporter to transform them to .x. There are thousands of models of all kinds of things available for SketchUp, so this way you can export them to .x and try loading them in the simulator. 

Best, 
Tanis

pattacini, 2016-03-01:
Thanks @tanismar for the link and the suggestions.

closed 2020-12-08 by pattacini