robotology/community #88: How to change F/T sensors offsets

hu-yue, 2016-01-18:
Hi, I have a lot of saturation issues with the F/T sensors, I remember there is a way to change F/T offsets which might help this, but I don't remember how and I can't find documentation about this. 
I found this http://wiki.icub.org/wiki/Firmware#Firmware_update_.28Force.2FTorque_Sensors.29, but it seems different from what I remember.
Thanks!

barbalberto, 2016-01-18:
Is it somehow related to this https://github.com/robotology/icub-support/issues/89 ?

hu-yue, 2016-01-18:
No, I mean, it's always during torqueBalancing, but somehow before Christmas break I did not have this issue anymore, now they are showing again. I already changed the saturation threshold from the configuration files in `/mechanicals`, as I mentioned in https://github.com/robotology/QA/issues/64, and the current limits as mentioned in https://github.com/robotology/icub-support/issues/90.

I remember seeing @francesco-romano opening some gui to change offsets, where it's possible to visualize also the calibration matrix, but I don't remember how anymore.

francesco-romano, 2016-01-18:
Hi @hu-yue.
Now that the firmware has the diagnostic set to on, it is easier to regulate the offsets.

- Make sure `robotInterface` is off
- launch the ethLoader and look for the boards
- close ethLoader
- launch `canLoader --calib`
- now you have to select the board. It should be on CAN2, and you should write the IP address of the board.
- It will open with a calibration matrix (don't touch it) and the 6 channels with relative sliders. You have to touch these sliders. Once you finish, press the `Save to eprom` button, close the window and reboot the boards.

hu-yue, 2016-01-18:
@francesco-romano thanks! =) I remember it was somewhere with `canLoader` but could not find the right place! Next time I'll take notes!

closed 2020-12-08 by pattacini