robotology/community #90: Get image of size 640*480 in iCub simulator

ZHUQINGPENG, 2016-01-25:
Hi all,

I want to get high resolution images (640*480) in the iCub simulator. In the website: http://eris.liralab.it/wiki/Simulator_README, it only talks about how to get the low resolution images.

- Cartesian ( 320 x 240 )
- Fovea     ( 128 x 128 )
- Logpolar  ( 252 x 152 ) 

I found a ini file named "icubSimEyes_640x480.ini", but I don't know whether it is related with my question or not.

Does any of you have idea to fix this?

Thank you very much.

Walter

vtikha, 2016-01-25:
Hi @ZHUQINGPENG , you just need to tell the simulator which configuration to choose at runtime. Look at the installed ``simulator.ini`` file. By default the camera file used is the ``icubSimEyes.ini``, replace that one with ``icubSimEyes_640x480.ini``to get the higher resolution cameras.

ZHUQINGPENG, 2016-01-25:
@vtikha It works well. Thank you very much.

closed 2020-12-08 by pattacini