robotology/community #91: Firmware update with ethLoader and canLoadeder

KevinAStein, 2016-01-27:
Hi,
we are just updating the firmware and were wondering if it is possible to upload the ems.hex for all boards in one go.
Same question for canLoader. Can we tick all boxes and upload (e.g. strain.hex)?
Thanks
@hu-yue

randaz81, 2016-01-27:
Hi, for the EMS yes, you can tick all the boards, since the firmware is the same (ems.hex) for all of them.
For the canLoader, it's possibile (in principle). However, execept for the strain.hex, all the other boards have different firmwares types.

KevinAStein, 2016-01-27:
@randaz81 thanks, but in the canLoader there is also only one 2FOC.hex, skin.hex and strain.hex file, is it not possible, for example on the board 10, all the boxes and load the skin.hex at once?

pattacini, 2016-02-12:
@randaz81 ?

julijenv, 2016-02-12:
yes you can load in one shot all the skin parts (with skin.hex)

pattacini, 2016-02-12:
Thanks @julijenv.

closed 2020-12-15 by pattacini