robotology/community #92: problem with icub_iso2usb.sh on Ubuntu 15.10

frank-foerster, 2016-02-01:
Hi all,

I hope I'm writing to the correct 'forum' - the last time I did iCub-related things, the robotcub-hackers mailing list was still fulfilling this purpose. So if this is the wrong forum, please let me know where to ask this question.

I'm currently upgrading our iCub's OS on the USB stick and found that when using the icub_2usb.sh script, the ensuing OS would not be bootable. I'm using Ubuntu 15.10.
If I use dd instead (so without persistence partition), the image works and I can boot from the USB-stick afterwards. 
The script doesn't output any error, but it 'says' at some point that it can't create a 744MB partition, but only a 780 one and asks for my confirmation. I answered the question with 'yes', but when I look at the created partition sizes afterwards the created partition seems to be 744MB large.
The output/error, when trying to boot, is simply:
"SYSLINUX 6.03 EDD20150813 Copyright (C) 1994-2014 H. Peter Anvin et al
Boot error".

Does anyone have an idea what could cause this? 

I did notice that the ensuing file system types of the Debian partition are different when using icub_iso2usb.sh as compared to using dd. Using the script the first partition is a Win95 fat32 partition
according to fdisk. Using dd, the file system type of /dev/sdc1 (in my case) is 'Hidden HPFS/NTFS'

Many thanks

Frank

pattacini, 2016-02-01:
Hi @frank-foerster 
This is the correct place to ask such a question :smirk: 

I think @mbrunettini might give you some help on this.

mbrunettini, 2016-02-02:
Hi,
as described in the instructions [here](http://wiki.icub.org/wiki/The_Linux_on_the_pc104#How_to_use_the_icub_iso2usb.sh_script), the script has been tested only on Ubuntu Trusty and Debian wheezy. 
This is the only working method to put the iso on the USB, please do not use other ways.
Moreover, please read all the instructions carefully.
Let us know.

frank-foerster, 2016-02-02:
Hi, 
I read all the details but thought I tried it with my current Ubuntu nevertheless. 
I'll give it another try with a Debian 7 and will let you know how that goes.

Thanks

Frank

kt10aan, 2016-02-02:
@mbrunettini, since the icub2iso is the only working method to put the ISO on the USB stick, and since Wheezy is oldstable for 9 months now, it would be nice if Debian 8 Jessie was added to the list of supported platforms.

mbrunettini, 2016-02-02:
You can use also Ubuntu Trusty. 
Unfortunately, the point that the script is not working on Debian 8 depends from diffences in the syslinux package, which is not under our control.

Tobias-Fischer, 2016-02-02:
@mbrunettini Does that mean that the script won't work with the upcoming Ubuntu 16.04 either, as the same `syslinux` version as in Debian Jessie is used? I guess that would be a big issue. How did other pieces of software which do a similar job solve the change in the `syslinux` package (`unetbootin` etc.)?

Tobias-Fischer, 2016-02-02:
Also see https://github.com/robotology/QA/issues/82

mbrunettini, 2016-02-02:
@kt10aan Please let us know if using a supported distro solved this issue.
Thanks

kt10aan, 2016-02-02:
@Tobias-Fischer +1

frank-foerster, 2016-02-03:
Hi, I just repeated the procedure with a Debian 7 distro and it worked with that one.

closed 2020-12-15 by pattacini