robotology/community #93: How to check if F/T sensors readings are correct

KevinAStein, 2016-02-02:
Hi,
we are wondering how to check if the readings of the F/T from the `analog:o` are correct?
For example putting the robot in upright standing position on the ground, we see that the readings from `right_foot\analog:o` and `left_foot\analog:o` are pretty different, while we thought they should be pretty similar.

Thanks!
cc @hu-yue

traversaro, 2016-02-02:
Hi @KevinAStein , consider that the measurements of the F/T sensors (as streamed by the `/icub/***/analog:o` ports ) are affected by an unknown offset. This offset is computed at the startup (and when the `calib***` rpc function are called) by the `wholeBodyDynamics` and `wholeBodyDynamicsTree` modules. 

To get some ports streaming measurements of the F/T sensors with this offset removed, launch the `wholeBodyDynamicsTree` module with the `output_clean_ft` option (you can simply add `--output_clean_ft` in the wholeBodyDynamicsTree option box in yarpmanager). 

Once you get the measurement without the offsets, you can try to load the robot with some additional object with a known weight to check the measurement of the F/T sensors (the use of an **additional** weight is necessary because the known weight of the robot is used to estimate the F/T offset).

Consider that for two feet balancing scenario, this offset is computed when you launch the `twoFeedStandingIdleAndCalib.sh` script, as described in https://github.com/robotology/codyco-modules/tree/master/src/modules/torqueBalancing . 

Furthermore, consider that apparently the offset computed for the F/T sensors in the lower limbs seems to be highly dependent on the load on the legs (i.e. if the robot is in single support on the left/right leg, in double support or with the root_link fixed to the pole). We are currently in the process of investigating this phenomenon with @fjandrad , for more info on this please check https://github.com/robotology/codyco-modules/issues/160 .

KevinAStein, 2016-02-02:
@traversaro thanks, but after launching the WBD3 with `output_clean_ft` how do we read the data without offsets? Is there a port from WBD3 where to read them?

KevinAStein, 2016-02-02:
@traversaro just found it, sorry, was not seeing it because of a typo in the paramters

KevinAStein, 2016-02-02:
@traversaro if in zero upright standing position my filtered readings are 
right_foot
`2.409474 0.740165 -121.195665 -0.276714 0.25103 0.133824`
left_foot
`-2.026843 1.423318 -129.524288 -0.42297 0.26009 -0.047352`
Would this be reasonable? The difference on the z axis is almost 10, which seems quite big, but maybe it's not?

traversaro, 2016-02-02:
I think that are reasonable values. At least they are similar to measurements that we get on working iCub when we do double balancing demos (cc @DanielePucci for confirmation). 

For the sake of completeness, consider that FT sensors are always affect by termal drift, so if you got the measurement some time after the calibration, this could also affect how much the sensor output is reliable.

hu-yue, 2016-02-02:
Ok thanks, so 10 is not a big gap I assume.
We took the measurements right after the calibration, so I hope there was still not much thermal drift?
We also tried to load some weights on the robot handles and the measurement seems reasonable so I guess the sensors are fine. I wanted to check them because the torqueBalancing is behaving somehow differently, but this is another issue I'll try to take care about.

I'll close this.

DanielePucci, 2016-02-03:
The difference in the readings looks reasonable to me.

closed 2020-12-08 by pattacini