robotology/community #94: Noise insulation for power supplies

frank-foerster, 2016-02-04:
Hi, 

I'm currently trying to get a quiet noise-reducing cabinet for the iCub power supplies.
We're still having the old XFR35-35 and XFR60-46 power supplies here and I just measured
a noise level of about 55dB with a smart phone app (not sure how accurate that one is). 
I was wondering if anybody had experience with this issue - I do know that the Osaka group are
using a quiet rack enclosure of a Japanese manufacturer and I have their details.
But obviously I need a European retailer and am currently speaking to the British 'The Rack People'. I'm considering the 4U Orion Acoustic Mini. 

So does any of the European iCub users have experience with noise insulation for their power suppies? 
And could anyone tell me
 - the average power consumption of the iCub when sitting / attached to its frame and moving the arms
  - the BTU rating (BTU = British Thermal Unit) of the power supplies. 

I searched online, but I could hardly only get info on the XFR35-35, but no BTU rating. 
Knowing the average and/or peak power consumption might be enough information for the retailer to give me qualified support. As I'm just ugrading our whole infrastructure I can't start it up and have a look at the display of the power supplies.

Many thanks.

Frank

lschilli, 2016-02-05:
A rack with some noise suppression is one thing. But a more silent power supply still makes sense. I do not know the cost of this or if it makes sense for your setup. But just in case it helps:
https://www.dropbox.com/s/5nd7451yui40ro7/2013-04-17%2013.26.41-1.jpg
This one is much more silent than the old power supplies.

frank-foerster, 2016-02-05:
Hi Lars, 
thanks for the hint. I had a quick look and prices of this one vary considerably depending on the maximum output wattage. Do you guys have one like that? If yes, what's your maximum wattage? I'll see if I can get any noise levels for it. The price for this one and the noise cabinet could nearly the same. So it really depends how quiet it really is.

lschilli, 2016-02-05:
This is the Osaka one. Maybe Kostas can look up the details. In general I suggest to check with icub-support first. In my view they are significantly more silent than the old ones. You will notice them probably a little on a video recording but they will not render the audio track close to unusable as before. Before you buy a silencing rack I would recommend to buy those first. If you put them in a silencing rack it will get even better but the next thing you will notice is the CPU fan of the pc104.

frank-foerster, 2016-02-05:
Ok, thanks. We're not doing audio recording at the moment but maybe in the future. Kostas already got back to me. So thanks again.

closed 2020-12-15 by pattacini