robotology/community #95: How to read power consumption of single motors

hu-yue, 2016-02-10:
Hi,
Is there a way to read/know the power consumption of each motor?
Accessing voltage/current readings?
Thanks!

pattacini, 2016-02-21:
Hi @hu-yue 

I think you should go as you were suggesting by getting the relevant quantities (voltage and current) by means of the motor interface API.

Be careful, because as far as I remember the current readings must be enabled from within the configuration file, at least for `CAN` robots.

hu-yue, 2016-02-21:
Hi, I was able to read the currents from the IAmplifier interface. At least, I've got some numbers out, I'm not quite sure how to interpret them though. I couldn't find it written in the documentation.

hu-yue, 2016-03-02:
I think we can close this.
I already know that current readings are in mA.

randaz81, 2016-03-02:
IAmplifierControl::getCurrent() returns the instantaneous current consumption in mA.
IAmplifierControl::getPowerSupplyVoltage() returns the power supply voltage of the board, expressed in Volts (work in progress)
The power consumption is thus P=V*I (expressed in mW). 
Currently you can assume V=40V (the voltage provided by the robot power supply) for all ETH boards.

closed 2020-12-08 by pattacini