robotology/community #96: Motor protection and current limits

hu-yue, 2016-02-10:
Hi,
In case we perform a motion (like stair climbing) with the robot that requires more current than the set limits, what exactly would happen in this case? Would the single joint go on `HW fault`?
How high would be the probability of damaging a motor despite the protection with the limits?
(with current limits I mean https://github.com/robotology/icub-main/blob/master/app/robots/iCubHeidelberg01/hardware/motorControl/left_lower_leg-ems7-mc.xml#L15 , is this correct?)
Thanks!

pattacini, 2016-02-21:
@randaz81 do you have some insights to share on that?

hu-yue, 2016-02-21:
Hi, I almost forgot about this issue. I've already performed the stair walking and the knee motor went HW fault due to overcurrent, I was able to perform the motion by increasing the current limit, but I'm still a bit concerned about how much it's safe to increase it and how fast does the limit trigger the HW fault? 
Thanks!

pattacini, 2016-03-24:
Better mark this issue as **unanswered** and close it.
@hu-yue feel free to ping again on this.

randaz81, 2016-03-25:
Current overload protection at firmware level is currently under refactoring/development.

_[...]but I'm still a bit concerned about how much it's safe to increase it and how fast does the limit trigger the HW fault[...]_ 

It is not safe. You be should be very careful until the new FW build is released.

closed 2020-12-15 by pattacini