robotology/community #97: Is "yarpmotorgui" an appropriate name for a joint interface?

DanielePucci, 2016-02-11:
Hi,
I was wondering if `yarpmotorgui` is the right name for an interface that is, at all effects and purposes, a joint interface.

I propose, `yarpjointgui` :)

@pattacini @randaz81 @vtikha

DanielePucci, 2016-02-11:
@iron76 @lornat75

traversaro, 2016-02-11:
I think no one in history was confused by this: joint are controlled by motors, and if you know the difference between joint-level or motor-level control you know so much about the yarp/icub infrastructure not to be confused by the name of the yarpmotorgui. 

I don't think that the trouble of renaming (and educating all the users about the new name/modifyng scripts and tutorials mentioning the old name) is worth.

pattacini, 2016-02-11:
I'm afraid @traversaro is right.

vtikha, 2016-02-11:
@DanielePucci you have a valid point, but @traversaro's answer could not have been any clearer.

francesco-romano, 2016-02-11:
```bash
if [ computer == Pucci's computer ]; then
  ln -s yarpmotorgui yarpjointgui
fi
```

naveenoid, 2016-02-11:
@francesco-romano :joy:

DanielePucci, 2016-02-11:
Bah, the name is semantically wrong, especially in light of coupled joints where motors and joints cannot be linked by a one-to-one relation during motion. But, it's up to you. It was just a proposal.

pattacini, 2016-02-11:
I'm afraid @DanielePucci is right.

pattacini, 2016-02-11:
(but we can keep the coupling secret :smirk:)

traversaro, 2016-02-11:
> Bah, the name is semantically wrong, especially in light of coupled joints where motors and joints cannot be linked by a one-to-one relation during motion. But, it's up to you. It was just a proposal. 

Let me amend my reply: 

> I think no one in history was confused by this: joint are controlled by motors, and if you know the difference between joint-level and motor-level control or about coupled joints, you already know so much about the yarp/icub infrastructure not to be confused by the name of the yarpmotorgui. 

More seriously, if we want to avoid confusion about joint/motor, I think improving documentation on this would be a much more efficient use of our time then changing the name and then manually update outdated documentation/code or educate users about the new name. 

Just to give a (clearly biased) example: I still type `robotMotorGui` most of the times I want to use `yarpmotorgui`, and I would literarly *hate* any upstream project that changes the name of one of its most used tools for two times in less then a year.

barbalberto, 2016-02-11:
:+1:  x Silvio :-)

DanielePucci, 2016-02-11:
Being conservative is the (other) root of all evil :D

 With that said, I understand your point.

pattacini, 2016-02-11:
@DanielePucci join the dark side

DanielePucci, 2016-02-11:
Unfortunately, I did :D

drdanz, 2016-02-11:
There was a discussion about the new name when we moved it in yarp, one of the proposed names was yarpjointmover. I still hate all the options, especially the name we choose because of the "gui" part.
Anyway we are stuck with this name for the time being... :scream:

closed 2020-12-15 by pattacini