robotology/community #98: Sampling eye-head coordination in the iCubs search space.

ashrafzia, 2016-02-11:
Hi Guys,

Am new to iCub. Please ignore any silly mistakes.

Am trying to **sample joint values** of **eye-hand coordination** in the iCub simulator.

Previously, i defined a position in the iCubs search space using the cartesian coordinate system, reach their using the right arm and when given the same x,y and z coordinates to the fixation point i can look at it and the head adjusts automatically.

Now, i want to **record** all values i.e joint angles of the right arm + wrist (**7 values**), pan, tilt and vergence of the eyes (**3 values**) plus pitch,roll and yaw of neck (**3 values**) when given a position (x,y,z coordinates) in the search space which is iteratively updated, so **3 values** for the coordinates.
**Total ( 7 + 6 + 3 = 16 )**

I have the following script to run my program:

```
#!/bin/sh
gnome-terminal --profile="Default" --tab -t "YarpServer" -e "bash -c \"yarp conf --clean; && yarp server; exec bash\"" \
--tab -t "iCubSimulator" -e "bash -c \"sleep 2 && iCub_SIM; exec bash\"" \
--tab -t "simCartesianController" -e " bash -c \"sleep 3 && simCartesianControl; exec bash\"" \
--tab -t "right_arm Controller" -e "bash -c \"sleep 4 && iKinCartesianSolver --context simCartesianControl --part right_arm; exec bash\"" \
--tab -t "iKinGazeCtrl" -e "bash -c \"sleep 2 && iKinGazeCtrl --from configSim.ini; exec bash\"" \
--tab
```
I can see the arm configuration when i move to a particular location, in a terminal which i have associated with the right_arm as:

```
Target rxPose   [m] = -0.400000	-0.400000	 0.800000	-0.350747	-0.897678	-0.266741	 1.314759
  Target txPose   [m] = -0.220152	-0.006505	 0.342514	 0.463208	-0.884254	 0.059442	 1.978500
Target txJoints [deg] = -95.000000	 23.780724	 61.731054	 82.149945	 40.205303	-11.674962	 6.022538
  norm(rxPose-txPose) = pos [m]: 0.629664
                        ang [*]: 1.09984
    computed in   [s] = 0.042202

```

**Question 1:** How can i access or store the above joint angles? that are opened in a separate terminal.

**Question 2:** What if i define a starting point and a final point in the search space for the arm to move around and that should be followed by the head (eyes+neck)?

**Question 3:** If the given points are unreachable then what value will be stored?

I have been using the following functions previously:
getPose() -> getting initial configuration of arm
gotoPoseSync() -> moving to a particular location in space along with the acknowledgement.
getFixationPoint() and lookAtFixationPoint() -> for the eyes.

Any guideline or comments if i am going in the right direction?

Thank you

pattacini, 2016-02-12:
Hi @ashrafzia 

Let me suggest you to use [**`yarpmanager`**](http://www.yarp.it/yarpmanager.html) to launch modules: it's much more handy than relying on dedicated scripts.

##### Answer 1
- You can access robot joints (e.g. neck, eyes, arm ...) through the `yarp::dev::IEncoders` motor interface. See this [**tutorial**](http://wiki.icub.org/iCub/main/dox/html/icub_motor_control_tutorial.html) to find out more.
-  You can easily log and store joints values as they change throughout your experiments relying on [**`yarpdatadumper`**](http://www.yarp.it/group__yarpdatadumper.html).

##### Answer 2
Perhaps I didn't catch your exact point, but the Cartesian space (what you call the "_search space_") is common to both the Cartesian controller of the arm and the Gaze controller; therefore, you can provide them with the same targets and, as result, the head and the eyes will somehow follow the arm movements.

##### Answer 3
Arm and Gaze controllers do employ nonlinear optimization underneath that intrinsically deals with unfeasible goals. Thus, if you ask to attain out-of-reach targets, the optimizer will instruct the controller to move to the closest point still reachable.

ashrafzia, 2016-02-15:
Hi @pattacini 

Thanks for your reply and guidance. 

I launched the **yarpmanager** but could not able to run sample application. I copied the template  iCub_Simulator_Startup and saved as my application file because it consists of all the required modules that am going to use. 

I get the following error:
`[ERR] (iCub_Simulator_Startup) cannot initialize broker. : console does not exist. check yarprun is running as server.`

I started [yarprun](http://www.yarp.it/yarprun.html), refreshed application in yarpmanager and finally restarted yarpmanager but still same error.

pattacini, 2016-02-15:
The `xml` of the application you want to launch via **`yarpmanager`** needs to contain the references to the `yarprun` nodes where those modules composing the application are meant to run.

In a very minimal setting where you launch every module on your local machine, you could do the following:

1. run `yarprun --server /my-pc` on your machine.
2. fill in the `node` tag in the `xml` file with `my-pc`.
3. run `yarpmanager` and open the modified `xml` file.

In the same scenario, you can profitably rely on the following shortcut, skipping the first step:

1. fill in the `node` tag in the `xml` file with `localhost`.
2. run `yarpmanager` and open the modified `xml` file.

Either way, you should be ready to run the application at this point.

If you will still have problems with `yarpmanager`, to keep this issue on track, please open up a dedicated ticket here: https://github.com/robotology/yarp/issues.

ashrafzia, 2016-02-15:
Thank you @pattacini 

I have managed to successfully run the module, but the **yarpdatadumper_recording_example** isn't working. [Here](https://github.com/robotology/yarp/issues/684) is the new issue.

Even if i manage to run data dumper the values for connected elements like head, torso, rightArm will be recorded but how can i track values for (Pan, Tilt and Vergence) of the eyes?

pattacini, 2016-02-15:
The head part encapsulate both the neck and the eyes encoders according to the following format:

|  neck  | neck  | neck  |  eyes  |  eyes  |  eyes  |
| :---: | :---: | :---: | :---: | :---: | :---: |
| tilt | roll | pan | tilt | pan | ver |

pattacini, 2016-02-18:
Closing; feel free to reopen if needed.

traversaro, 2016-02-18:
I would add to the table of @pattacini that the axis names used in the iCub software (for example the names returned by the [`IAxisInfo`](http://www.yarp.it/classyarp_1_1dev_1_1IAxisInfo.html) interface or the one used in URDF/SDF files) are [1] : 

|  neck  | neck  | neck  |  eyes  |  eyes  |  eyes  |
| :---: | :---: | :---: | :---: | :---: | :---: |
| `neck_pitch` | `neck_roll` | `neck_yaw` | `eyes_tilt` | `eyes_version` | `eyes_vergence` |

@pattacini I took the liberty of adding this line to your table. 

[1] : http://wiki.icub.org/wiki/ICub_joints#Head_2.0

pattacini, 2016-02-18:
Great :+1:

closed 2020-12-08 by pattacini