robotology/community #99: By gregoire-pointeauIs:  there a way to introduce delay in a connection ?

apaikan, 2016-02-16:
By gregoire-pointeauIs 

Hello.
Here is my problem.
I have a module (let's call it A) that streams information to a second module (let's call it B).
I would like to introduce some delay between the A and B.
My idea would be to create a module C in the middle, that would get the Bottle from A, wait, and forward it to B.
I wonder if something like that already exists in port monitor or something equivalent ?

Thank you in advance !

apaikan, 2016-02-16:
Hi @gregoire-pointeau . 
Yes! the portmonitor is the correct way to do and here there is an example what exactly what you need: 

https://github.com/robotology/yarp/blob/master/example/portmonitor/port_rate/rate.lua

here is also documentaion regardin the portmonitor if you have not been used it yet: 
http://www.yarp.it/portmonitor.html

gregoire-pointeau, 2016-02-18:
Hi @apaikan , thank you for your answer.
I'll have a look at it, and I think come back to you soon!

pattacini, 2016-02-25:
For the time being we can close this.
@gregoire-pointeau feel free to open it again when deemed necessary.

closed 2020-12-08 by pattacini