robotology/community #100: NEI - how to use to measure deflection of the spring

hu-yue, 2016-02-25:
Hi,
as we broke some wires as reported in https://github.com/robotology/icub-support/issues/171, now I discovered that there is this device which should measure the deflection of the spring.
However, I have no idea about how to use it?
As we keep breaking these wires (during walking, squatting, basically whatever requires large knee motion or during insertion/removal or the spring), if there's no way to use it for the time being, I would stop repairing it (for now).
Thanks!

pattacini, 2016-03-04:
cc @robotology/icub-support-team

pattacini, 2016-03-10:
let's try again with @randaz81 @maggia80 :smirk:

pattacini, 2016-03-24:
Better mark this issue as **unanswered** and close it.
@hu-yue feel free to ping again on this.

hu-yue, 2016-05-23:
Hi,
I would like to reopen this issue as we have now a student working on the springs of the robot, we'd like to know how to deal with this sensor? @randaz81 @maggia80 
Thanks!

randaz81, 2016-05-23:
Currently ETH firmware is not able to read NEI sensor. I'm thus transforming this issue into a feature request @valegagge @marcoaccame

traversaro, 2016-05-23:
Probably also @TeoKa is interested in this.

pattacini, 2016-05-29:
Hi @randaz81 

Can we close this issue here and open up a **feature request** within the ticket system of proper repository? I think it'd be a cleaner solution. You could then reference this issue from therein.

pattacini, 2016-06-07:
Ping @randaz81

pattacini, 2016-06-08:
Issue moved to [robotology/icub-firmware #69](https://github.com/robotology/icub-firmware/issues/69) via [**ZenHub.io**](https://www.zenhub.io/)

closed 2020-12-08 by pattacini