robotology/community #101: Find the DH parameters for ICub Simulator Arm

LakshithaHKUST, 2016-03-07:
Hello Everyone,

I would like to know whether there is a way to find the DH parameters for the ICub simulator arm?
There seems to be different versions of the arm and I am little bit confused.

Thanks in advance.

pattacini, 2016-03-07:
As per the [`code`](https://github.com/robotology/icub-main/blob/master/src%2Fsimulators%2FiCubSimulation%2Fodesdl%2FiCub.cpp#L3982), the kinematic version used by the simulator is the default one, that is `1.0`.

pattacini, 2016-03-07:
To retrieve the corresponding DH parameters, you can refer to our [**wiki**](http://wiki.icub.org/wiki/ICubFowardKinematics_left).

LakshithaHKUST, 2016-03-07:
Thank You very much for the prompt reply. I'll proceed with v1.0 DH parameters.

pattacini, 2016-03-07:
@LakshithaHKUST 

You should have received an invitation to join our developers' group.
Once accepted you'll be entitled to assign **labels** to issues herein.

LakshithaHKUST, 2016-03-07:
could you please send me the invitation again, my email had a typo
sorry about the inconvenience. thanks for adding me to the developers' group.

pattacini, 2016-03-07:
Sure; just done!

closed 2020-12-09 by pattacini