robotology/community #103: Announcing incompatibilieties and changes in YARP/iCub software

kt10aan, 2016-03-14:
I had talked about this a long time ago, maybe before the QA repo was started. But I think it is a good time to bring it up again.

The YARP/iCub software is continuously evolving and quite often changes break older stuff. It can be innocent renames in the API (like moving math functions from ``iCub::ctrl`` to ``yarp::math``, it can be renaming and/or moving of essential modules ``iCubInterface`` to ``robotInterface`` to ``yarprobotinterface``, it can be a new format of config files etc.

While I do keep up with all the posts in the QA/yarp/icub-main etc, many people do not or cannot, for various reasons. Also, despite keeping up with all the posts, I do find it difficult to remember when a breaking change was announced, and in which repository and go back to search for it.

I think it would be very nice if there was (for example) a wiki page that would list any incompatibilities or changes introduced (not for ones already happened but for future ones).

In my opinion, that would make the life of people outside IIT much much easier during upgrades and during updating older code.

At the same time, it will save support time, as quite a few of the support requests have to do with changes that have been introduced and the user was unaware of.

pattacini, 2016-03-14:
Hi @kt10aan 

Yours is a very good suggestion we are already trying to target with our brand new policies for the workflow. These policies will be out soon, hopefully, including a proper way to make very clear what changes will break compatibility needing all developers to be very much aware of.

Stay tuned.

kt10aan, 2016-03-14:
@pattacini great news!

pattacini, 2016-03-16:
@kt10aan, if you want to have already a "sneaky" preview of what these policies will be about, take a look at these [**instructions**](https://github.com/robotology/yarp/blob/master/.github/CONTRIBUTING.md).

With regards to the list of **new features** and **bug fixes** published upon taggin the stable branch - comprising breakage of back-compatibility -, we are thinking of exploiting directly the way Github allows detailing software releases.

pattacini, 2016-03-24:
Closing for now.

lornat75, 2016-03-25:
These are the pages that list the changes introduced at each past and upcoming releases:

iCub: http://wiki.icub.org/wiki/ChangeLog

YARP:  http://www.yarp.it/changelog.html

@drdanz @randaz81

closed 2020-12-08 by pattacini