robotology/community #104: How to I stream the output of a webcam on a YARP port?

traversaro, 2016-03-16:


traversaro, 2016-03-16:
If your webcam is supported by OpenCV, you can easily stream it on a YARP network using the `opencv_grabber` device. [1] [2]

First, make sure that you compiled YARP with OpenCV support, by opening making sure that the `YARP_USE_OPENCV` CMake advanced option is setted to `ON`.

Then you can check if the `opencv_grabber` device is available on your machine by using the `yarpdev --list` command. 

If it is not available, you should enable its compilation in YARP by enabling the `CREATE_DEVICE_LIBRARY_MODULES` CMake option and then the `ENABLE_yarpmod_opencv_grabber` CMake option. After you enabled it, compile (and install, if you usually do install) YARP. Now you should find the `opencv_grabber` when you type `yarpdev --list`. 

Now you can launch the device by using the following command:
~~~
yarpdev --device opencv_grabber
~~~

Now the video from the webcam should be streamed on the `/grabber` yarp port, but you can change the name of the port opened using the `--name` command line option. 

The video streamed can be easily dumped to a mkv video file using the [`yarpdatadumper`](http://www.yarp.it/group__yarpdatadumper.html) `–addVideo` option. [1]
 
[1] : http://www.yarp.it/group__opencv__grabber__basic.html
[2] : http://www.yarp.it/group__cmd__device__opencv__grabber.html

cc @fjandrad

dcam0050, 2016-03-16:
install yarp with opencv_grabber

then do yarpdev --device opencv_grabber \nameOfPort

drdanz, 2016-03-16:
Another option is to use the `ffmpeg_grabber` device:

```
yarpdev --device ffmpeg_grabber --v4l2
```

drdanz, 2016-03-22:
Yet another option... on linux the usbCamera device (from the [robotology/icub-main](https://github.com/robotology/icub-main/) repository) uses video4linux and therefore can be used to stream from a webcam:

```
yarpdev --device grabber --subdevice usbCamera --d /dev/video0
```

Thanks @barbalberto for this hint.

closed 2020-12-08 by pattacini