robotology/community #105: Attaching portmonitor scripts with specified parameters

pattacini, 2016-03-17:
Hi @apaikan 

Do you know whether there's the possibility to specify a parameter in the syntax we use to attach a lua script to a connection?

@Tobias-Fischer is interested in this.

apaikan, 2016-03-29:
Passing the parameters to a portmonitor plugin (lua/C++) at the connection time is not supported yet due to the limitation imposed by YARP carrier syntax. However, the paramters can be set/configured after establishing the connection via yarp administrative commands: 

- See [An example which shows how to use a Lua script to modify image data] (http://www.yarp.it/image_modification.html)

There are also some predefined attributes which can be accessed from the plugin scripts such as 
- `source` : the source port name 
- `destination`: the destination port name 
-  `carrier`: the full carrier string 

These are formatted as the YARP property and given to the portmonitor create method: 

```lua
-- create is called when the port monitor is created 
PortMonitor.create = function(options)
    local sourcePort = options:find("source"):asString()
    print(sourcePort)
    return true;
end
```

Similarly the `carrier` attribute holds all complete carrier parameter used for connecting ports: 
For example, if a lua plugin attached using the following command:
```
$ yarp connect /write /read tcp+recv.portmonitor+type.lua+file.codec+myparam.foo
```
Then the `carrier` attribute holds the  `tcp+recv.portmonitor+type.lua+file.codec+myparam.foo` string.  Therefore a workaround will be to pars this string within the lua code to extract the user-defined parameter (`myparam = foo`): 

```lua
PortMonitor.create = function(options)
    local carrier = options:find("carrier"):asString()
    print(carrier)
    for attr in string.gmatch(carrier,'([^+]+)') do
        k, v = attr:match("([^.]+).([^.]+)")
        print(k,v)
    end
    return true
end
```

You will get: 
```
t	p
recv	portmonitor
type	lua
file	codec
myparam	foo
```

Ignore the `t    p` ;)

Tobias-Fischer, 2016-03-29:
Thanks @apaikan! That's great. I'll give it a go soon.

pattacini, 2016-03-29:
Thanks @apaikan for the suggestion!
Closing for now; @Tobias-Fischer feel free to post here again.

closed 2020-12-08 by pattacini