robotology/community #106: Upgrade control boards firmware on Vizzy

plinioMoreno, 2016-03-31:
@randaz81,

We want to update to the latest yarp and iCub repositories, and we want to know if we need to upgrade the firmware of our control boards. The current versions as follows:

RM 4DC 1.11.82 (1.17.82)
RM BLL 1.61.82 (1.97.82)
Eyes: RM 4DC 1.15.80 (1.21.80)

And I would like to know iwhat would be needed to use the saccadic mode on Vizzy (and if it's possible at all).

Thanks in advance

randaz81, 2016-04-01:
Hi @plinioMoreno,
current firmware build is 84. I'm not aware of any compatibility issues, you can easily update the boards using the binaries available in the git repo **icub-firmware-build**. Remember: the first two numbers must NOT change (e.g. for the eyes board use the binary for version 1.15 and NOT the one for version 2.15).
Remember also that, besides **yarp** and **icub-main**, you also need to update **icub-firmware-shared** repo. It may happen that some new parameters have been recently aadded to the xml files. In case of problems, just ask: we'll add the correct parameters for your robot.

Regarding your questions about saccadic mode, the control is entirely handled by **iKinGazCtrl** module (@pattacini), not by the firmware. At the firmware level, the joint is controlled in **mixed mode**. This feature is available since firmware build 60, so your boards are already capable of it.

plinioMoreno, 2016-04-01:
Hi @randaz81,

Thanks for the info. I'll contact you in case the xml files need to be changed.

@pattacini, I'll update Vizzy's control gaze code and let you know about the saccadic mode.

closed 2020-12-15 by pattacini