robotology/community #107: Mjpeg carrier gives error

muhammadalidildar, 2016-04-25:
Hi all! I am trying to transfer iCub cam images to a non-YARP application. Here I must mention that I didn't install YARP and iCub from sources. I installed from executables on Windows. Now I am trying to transfer images using mjpeg carrier but the errors give me impression that the required libraries are not installed. If so, please help me how can I tell YARP to enable support for the carrier mjpeg?

This is what I did to test this connection on my pc,
`yarpdev --device test_grabber --name /grabber`

[image]

`yarpview --name /view`

Getting error in response to the connection
`yarp connect /grabber /view mjpeg`

[image]

For my satisfaction I tried to get images in firefox,
[image]

and here is the error,
[image]

Thanks in advance.

drdanz, 2016-04-26:
Unfortunately, the mjpeg carrier is not shipped with the binary packages yet.
At the moment the only way to get mjpeg working is to build it from sources.

Anyway, since the plugin system should be working properly, I think that we can try building it and sending you the dll, but you need to let us know which version of MSVC you downloaded, and which architecture (32 or 64 bit).

muhammadalidildar, 2016-04-26:
Thank you very much for your kind favor. I installed YARP for Visual Studio v11 (2012), 32 bit.

drdanz, 2016-04-28:
@muhammadalidildar I think this should be all you need to get it working, but I couldn't try it... [mjpeg_carrier_v11_x86_0.zip](https://github.com/robotology/QA/files/240153/mjpeg_carrier_v11_x86_0.zip)

You have to extract the dll's in the plugin folder (that should be something like `C:\Program Files\robotology\yarp\lib\yarp`). I'm not sure if there is a `Debug`/`Release` somewhere (either before `lib`, or before or after the second `yarp`. Eventually you have to put

* `yarp_mjpeg.dll` in the Release folder
* `yarp_mjpegd.dll` and `yarp_mjpegd.pdb` in the Debug folder

Otherwise just put all the files in the same folder.

You should already have a file `mjpeg.ini` in `C:\Program Files\robotology\yarp\share\yarp\plugins`, if it is missing, you can download it from here: https://raw.githubusercontent.com/robotology/yarp/v2.3.64/src/carriers/mjpeg_carrier/mjpeg.ini

You should then be able to test if the plugin is working by running on a terminal `yarp plugin mjpeg`.

If you don't get errors, then everything should be working properly, and you can try again with the connection.

Let me know if you manage to get it working (also if the paths are correct).

muhammadalidildar, 2016-04-28:
Hi! I just placed all the three files in `C:\Program Files (x86)\robotology\yarp-2.3.64\lib\yarp` and for the .ini file, I already had it in the right place. Now its working fine.
I am very much thankful and satisfied with your kind support.
Thanks once again.

closed 2020-12-15 by pattacini