robotology/community #108: Create custom configuration file with home position for the yarpmotorgui

UChervet, 2016-04-26:
Dear all,

I freshly installed yarp iCub and the codyco superbuild to work with the WBI and Gazebo.
When I try to bring the robot to an home position with yarpmotorgui I get an error : 
"Asking the yarprobotinterface to homing part right_arm through the remoteCalibrator interface, since no custom zero group found in the supplied file."

How do I create this custom zero group ? Is there a configuration file that I have to create ?

Thanks for your help !

jeljaik, 2016-04-26:
There is already one in `/icub-gazebo/conf`. First `cd` to that location and then launch the `yarpmotorgui` as: `yarpmotorgui --from robotMotorGui.ini`

pattacini, 2016-04-26:
@jeljaik I feel to really encourage the use of the command line option `--context` rather than `cd`-ing straight in the directory :wink:.

UChervet, 2016-04-26:
@pattacini How should I call it with the context option ?

pattacini, 2016-04-26:
@UChervet 
```sh
$ yarpmotorgui --context context-name --from robotMotorGui.ini
```
where **`context-name`** is the name of the context where you installed the bundle of `icub-gazebo` applications.

See #42 for more details on this topic.

UChervet, 2016-04-26:
Thanks I put  my custom configuration file in my YARP_DATA_DIRS personal folder and I works even without context.

barbalberto, 2016-04-26:
Dear all, home and calibration parameters **SHOULD NOT** be inside the motorgui config file.

Instead, the calibration and homing are remote, in the sense that values are stored by the yarprobotinterface and the motorgui only ask to perform the operation, without providing any parameters.
If it does not works, it means the config file of the robot is missing the following part:

in file <yourRobot>/wrapper/motorControl/<partName>_mc_wrapper.xml
add the ```Calibrator``` tag.

This tag tells the wrapper which calibrator it has to use when homing/calibrating
for example for torso
```
      <action phase="startup" level="5" type="attach">
      <paramlist name="networks">
         <elem name="torso_joints">  torso_mc </elem>
	 <elem name="Calibrator">  torso_calibrator </elem>
      </paramlist>
      </action>
```

This has been done to avoid duplication of sensitive calibration parameters in different files and having to customize too much the motorgui config files. 
Values in the config file are still supported as a fallback in case the wrapper is not yet configure as shown before (and/or to have configuration that may differs from the standard one, but they are discouraged).

barbalberto, 2016-04-26:
Sorry, I just read this was about the simulator... I was speaking about the real robot.

Since the simulator does not support calibration device yet, the zero values must still be in the motorgui config file.

An improvment could be to introduce calibrationg, at least homing/parking, device in the simulator to have the same behaviour.

randaz81, 2016-04-26:
> Since the simulator does not support calibration device yet, the zero values must still be in the motorgui config file. 

But please consider this just a temporary fix... Home values will be remotized soon :)

closed 2020-12-15 by pattacini