robotology/community #110: a question & images regarding  icub skin

frank-foerster, 2016-04-29:
Dear all,
I have been contacted by an content developer from the Science museum in London with regards to the iCub skin. As I know next to nothing about its skin I decided to ask the wider iCub community, that is you, for help. 

Firstly, she wants to know whether the patch of skin on the attached image is identical to the one employed on the iCub. 
So my first question: is the skin used on the iCub identical to the one visible on the attached image?
(This is assuming that one can tell by mere visual inspection. If this is impossible to tell, this information might also be helpful for them.)

Secondly, she was asking whether there were any images of a 'stripped down' iCub, that is, an iCub where its skin sensors are exposed. 
They would like to use such an image in their upcoming robot exhibition. The purpose of this image would be to give visitors a better idea of how such skin is employed on an actual robot, and what the difficulties might be such as potential problems with pliability if one were to add skin on the 'elbow' or other more 'bendy' parts of a robot etc. 
So does anyone have images of an 'exposed' iCub, optimally in high quality which, they would be willing to 'borrow' to the museum?
If not I will do the 'stripping' myself, but if anyone already has such images, it would save me some time.

Thanks.

[image]

pattacini, 2016-05-01:
Maybe @maggia80 can help in this respect.

maggia80, 2016-05-02:
Hi Frank,

No, this is not the iCub skin. I would like to know where she found it, if it is possible. Here in attachment the picture of the icub skin. I'll check if I can find also a picture of the skin on iCub.
[image]

maggia80, 2016-05-02:
few more picture 

https://www.dropbox.com/sh/qf4bb19xha1u0w9/AAB6O6ElGjWi2BemU8adXhaXa?dl=0

frank-foerster, 2016-05-02:
Thanks for the images - I'll forward them to Anna. If they like them, is it ok for them to use them in their exhibition?
I will also tell her that the two types of skin are not the same, ask her for some more detail, and will get back to you.

pattacini, 2016-05-02:
Hi @frank-foerster 

There's no problem in using the pictures in the exhibition.

alecive, 2016-05-04:
@matejhof should have a pretty good picture of the blue robot with all the skin parts exposed if I recall correctly!

frank-foerster, 2016-05-05:
Hi @maggia80, Anna finally got back to me. Evidently the piece of skin on the image is a loan given to them by Giorgio Cannata at the University of Genoa who were involved Roboskin (and are presumably around your corner). She sent me a link to a leaflet from Roboskin on which there are images of similar skin as the one on the image she sent. 
(http://cordis.europa.eu/docs/projects/cnect/0/231500/080/publishing/readmore/ROBOSKIN-Leaflet.pdf)
Maybe there has been some early version prior to the one used now? 
Anyways, she was asking me whether she can contact you directly as they would need higher-resolution images if available. Can I forward you email address to her? (As I don't have yours you'd have to send that one to me, my email address is on my github profile page).
@alecive: Thanks for the hint. I'll try to contact Matej directly. 

Thanks for all the help.

maggia80, 2016-05-05:
ok yes I realized that it could be a picture from Giorgio. give my contact directly to her

frank-foerster, 2016-05-05:
@maggia80: I actually don't have your email address (and it's not stated on your Github profile). 
Could you send it to me? (I put mine on the profile).
Thanks

maggia80, 2016-05-05:
I sent you my  mail

pattacini, 2016-05-05:
Closing here.
Feel free to reopen.

closed 2020-12-08 by pattacini