robotology/community #111: ETH robots: what is the mapping between the EMSs addresses and the robot associated parts?

DanielePucci, 2016-05-08:
Is there a documentation where it is explained the mapping between EMSs and robot associated parts? For instance, the EMS 6 is in charge of handling the motion control and the strain for the left upper leg. Can we have a complete description for the entire robot? @marcoaccame @valegagge @ale-git @randaz81

pattacini, 2016-05-10:
@marcoaccame are you willing to take on this?

marcoaccame, 2016-05-11:
Hi @pattacini, I can prepare the document. 

Where do you think it is best to publish it?

pattacini, 2016-05-11:
Thanks @marcoaccame 

Ideally, the place should be our [wiki](http://wiki.icub.org/wiki/Main_Page), but it's better to talk to @randaz81 for that.

DanielePucci, 2016-05-11:
👍

pattacini, 2016-06-03:
Hi @marcoaccame any update on this?
If the documentation is ready we could post a link to it here and close the issue.

pattacini, 2016-06-07:
Ping @marcoaccame

traversaro, 2016-06-07:
cc @fjandrad

marcoaccame, 2016-06-15:
The description of how the ETH robot is formed is now in http://wiki.icub.org/wiki/Firmware#ETH_robots

closed 2020-12-15 by pattacini