robotology/community #112: How do I read data from a BNO055 Xplained Pro board using a TTL-232R-3V3 cable and YARP?

traversaro, 2016-05-10:
Datasheet BNO055 : https://www.bosch-sensortec.com/bst/products/all_products/bno055 .
Docs Atmel BNO055 Xplained Pro : http://www.atmel.com/tools/ATBNO055-XPRO.aspx?tab=documents
TTL-232R-3V3 cable : http://www.ftdichip.com/Products/Cables/USBTTLSerial.htm

# Connect the  TTL-232R-3V3 to the BNO055 Xplained Pro pinout 
| TTL-232R-3V3  |  BNO055 Xplained Pro | 
|:--------------------:|:-------------------------:|
| RED         |  20 |
| BLACK    | 2    |
| YELLOW |  11 | 
| ORANGE | 12 |   

[image]

# Prepare YARP
Compile YARP with the `imuBosch_BNO055` device.

# Launch the driver 
~~~
yarpdev --device inertial --subdevice imuBosch_BNO055 --comport /dev/ttyUSB0 --baudrate 115200 --name seesaw
~~~

# Read the data 
yarp read ... /seesaw 

cc @DanielePucci @francesco-romano @nunoguedelha and kudos to @barbalberto @maggia80

traversaro, 2016-05-10:
Datasheet BNO055 : https://www.bosch-sensortec.com/bst/products/all_products/bno055 .
Docs Atmel BNO055 Xplained Pro : http://www.atmel.com/tools/ATBNO055-XPRO.aspx?tab=documents
TTL-232R-3V3 cable : http://www.ftdichip.com/Products/Cables/USBTTLSerial.htm

# Connect the  TTL-232R-3V3 to the BNO055 Xplained Pro pinout 
| TTL-232R-3V3  |  BNO055 Xplained Pro | 
|:--------------------:|:-------------------------:|
| RED         |  20 |
| BLACK    | 2    |
| YELLOW |  11 | 
| ORANGE | 12 |   

[image]

# Prepare YARP
Compile YARP with the `imuBosch_BNO055` device.

# Launch the driver 
~~~
yarpdev --device inertial --subdevice imuBosch_BNO055 --comport /dev/ttyUSB0 --baudrate 115200 --name seesaw
~~~

# Read the data 
yarp read ... /seesaw 

cc @DanielePucci @francesco-romano @nunoguedelha and kudos to @barbalberto @maggia80

maggia80, 2016-05-26:
actually the FTDI cable you are using is not correct since the Vcc is 5V, only the TX and RX has the 3V3 levels.

The right cable is the TTL-232RG-VREG3V3-WE

traversaro, 2016-05-26:
I check the document and the pinout should be consistent, however this still need to be tested.

closed 2020-12-08 by pattacini