robotology/community #114: Super-impose the kinematic model on the camera images

vicentepedro, 2016-05-15:
Hi all!

I was wondering if there is a YARP/ICUB module to super-impose the kinematic model of the iCub Arms (using iKin) on the camera images.

Thanks in advance,

pattacini, 2016-05-16:
Hi @vicentepedro 

As of now there's no proper module I'm aware of that does what you need. However, you could get inspiration from this [**code**](https://github.com/pattacini/icub-contrib/tree/master/show-fingers), which basically draws the fingers skeleton on the images acquired from the camera.

In your context, you might find useful to rely on the version of [**`getPose`**](http://www.yarp.it/classyarp_1_1dev_1_1ICartesianControl.html#a98b6776fcd71f7e3762bbc8c0afc0d42) that deals with the specific axis.

@vtikha and @claudiofantacci may want to further comment on this.

claudiofantacci, 2016-05-16:
Hi @vicentepedro 

as @pattacini said, [show-fingers](https://github.com/pattacini/icub-contrib/tree/master/show-fingers) is a valid starting point for your need.

You may also want to check some of the code I wrote starting from [show-fingers](https://github.com/pattacini/icub-contrib/tree/master/show-fingers), in particular the [`DrawHandSkeletonThread `](https://github.com/claudiofantacci/movefinger/blob/master/src/main.cpp#L177-347) class.
Please be aware that my code is still _under development_ and some errors may occur.

Should you require further insight, let us know.

vicentepedro, 2016-05-19:
Thank you, @pattacini and @claudiofantacci.

I tested the module (show fingers) and for now, I think it is sufficient for what I need.

pattacini, 2016-10-29:
@claudiofantacci and @vicentepedro you might want to consider info detailed in #152.

claudiofantacci, 2016-10-29:
Thanks @pattacini!

vicentepedro, 2016-10-29:
Thanks @pattacini!

closed 2020-12-08 by pattacini