robotology/community #115: iCub "falling" in Gazebo

apostroph, 2016-05-16:
Hi,

I am migrating from iCub_SIM to Gazebo, but the iCub appears to "fall" whenever I send a command Cartesian controller in Gazebo.

I am using the command icart->goToPose(xd, od); to set the right end-effector to:
xd = -0.4 : 0.1 : 0.2
and od = -0 : 0.707107 : -0.707107

It works very well with the iCub_SIM, but not with Gazebo (see enclosed screen-shots)

I tried to start the wholeBodyDynamics or/and the gravityCompensator with the "robot" variable set to  "icubGazeboSim", but it still does the same thing (aka falling down)

Any idea why this is happening?

Best,
Jimmy Baraglia

[image]
[image]

traversaro, 2016-05-16:
I think that the problem is that in Gazebo simulation, the velocity control is not properly tested/tuned (see https://github.com/robotology/gazebo-yarp-plugins/issues/240). 
This is happening because `simCartesianControl` is using velocity control by default, you can try to make it use the `POSITION_DIRECT` mode by setting the [`PositionControl`](https://github.com/robotology/icub-main/blob/master/app/simCartesianControl/conf/simCartesianLeftArm.ini#L7) option of the `simCartesianControl` to on.

apostroph, 2016-05-17:
I changed the option PositionControl to "on" in simCartesianRightArm.ini and simCartesianLeftArm.ini.
It worked perfectly, thank you.

Cheers,
Jimmy

tkelestemur, 2016-06-24:
I still have the falling problem even though I change `PositionControl` to on in both `simCartesianRightArm.ini` and `simCartesianLeftArm.ini`. I am running `tutorial_cartesian_interface` [from icub-tutorials](https://github.com/robotology/icub-tutorials/blob/master/src/motorControlAdvanced/tutorial_cartesian_interface.cpp). After two second running the code iCub falls backwards. 

[image]

traversaro, 2016-06-24:
By default no balancing controller is running on icub when simulated on Gazebo. To simulate the robot 
as if it was fixed on a fixed base, can you try to launch  the `iCub (fixed)` model? 
If that model is without hands we need to updated it, it is not a big problem.

tkelestemur, 2016-06-24:
Thank you very much @traversaro, with `iCub (fixed)` it is working now. I wonder why same code causes falling in Gazebo but not in default simulator? From your point is some kind of balancing controller running when we run `iCub_SIM`  ?

pattacini, 2016-06-24:
@tkelestemur there's no balancing in `iCub_SIM`, and be also careful because `fixed_hip` is `on` by default in `iCub_SIM`; that's why usually you don't see any fall of the iCub therein.

That being said, it would be normal any how to get different results with different simulators.

closed 2020-12-09 by pattacini