robotology/community #116: Sending rpc command via code

KevinAStein, 2016-06-06:
Hi,
I wanted to send commands to open rpc ports using C++.  I followed the client example from http://www.yarp.it/rpc_ports.html , but I don't manage to get the same result as when and using yarp rpc manually in a terminal.
thanks,
Kevin

towardthesea, 2016-06-06:
Hi @KevinAStein , could you please provide more details how you did it and the response, e.g terminal snapshot. Without the detail, I am afraid that nobody can give you advice!

Regards,
Phuong

KevinAStein, 2016-06-06:
Hi,
in Terminal I use "yarp rpc /portname/rpc:i" and then send my command to the module. 
Instead I want to use something like:
```
  RpcClient port;
  port.open ("/app/Port:o");
  Network::connect("/app/Port:o", "/portname/rpc:i");
  Bottle cmd;
  cmd.addString("do something");
  Bottle response;
  if (port.write(cmd, response)){
    cout << response.toString() << endl;
  }
```

If I use the help command I get the same output. But using other commands it seems that the code version is not waiting for the module to finish the task and instantly responds in failure.

towardthesea, 2016-06-06:
Hi @KevinAStein ,

From the above mentioned code, I only can say that your created `client` will open an `rpc port` and then connect to the the `/app/Port:o server port`. Then, it sends a `do something` string to the `client port` and display the response from the `server`. If it is your expected result, you got it. 
Actually I am afraid I cannot say more without knowing what did you do with your `server`!

P/S: I assume that you followed the  tutorial at http://www.yarp.it/rpc_ports.html and got the same results!

Regards,
Phuong

KevinAStein, 2016-06-06:
Yes, I understand what my code is supposed to do, but why is there a difference between running it manually in the command line interface or within C++? Am I missing something?

pattacini, 2016-06-06:
Hi @KevinAStein 

Would you mind to post your code, both the client and the server part, using [**`Gist`**](https://gist.github.com)? This way, we'll be able to compile and run it and see what's going on.

KevinAStein, 2016-06-06:
I am trying to interact with WholeBodyDynamicsTree for the icub robot. I want to send the calibration command when initializing my module instead of doing it manually in a terminal everytime. There is only the lines posted above.

pattacini, 2016-06-06:
Then, it might be a sort of mismatch between the sent command's format (a string?) and the expected one.

In `ARE` we reset `WBD` using this [piece of code](https://github.com/robotology/icub-main/blob/master/src/modules/actionsRenderingEngine/include/iCub/MotorThread.h#L267) and it's working fine (disregard the call to `Time::delay` since it has nothing to do with the rpc comm).

However, I'm not into the `WBDTree` rpc format: I'd suggest you to check that out, therefore.

pattacini, 2016-06-06:
One pitfall could be also that `addString("do something")` is different from `addString("do"); addString("something")`. Don't know if this applies to your case though.

KevinAStein, 2016-06-06:
thanks, that actually solved my problem!

pattacini, 2017-01-10:
cc @Iaxama

closed 2020-12-15 by pattacini