robotology/community #117: Problems with icub-live 6.0 image

jeljaik, 2016-06-07:
Since we were recently updating everything on iCubParis02 we noticed there was an icub-live 6.0 and decided to give it a try. The first thing we notice is that the `.iso` was no longer provided but we created it by doing: `genisoimage -o output_image.iso directory_name`. The USB key was totally formated and has one single partition. Nevertheless, when using the `icub_iso2usb.sh` script we get the following error after some time: 

```bash
icub@icub-desktop:~/Downloads/pc104-images$ sudo ./icub_iso2usb.sh -f icub-live_6.0-16.04.05-rt-amd64.iso -t sdb
checking mkfs.ext4 executable..
/sbin/mkfs.ext4
checking install-mbr executable..
/sbin/install-mbr
checking syslinux executable..
/usr/bin/syslinux
checking 7z executable..
/usr/bin/7z
checking mkdosfs executable..
/sbin/mkdosfs
checking mkdiskimage executable..
/usr/bin/mkdiskimage
checking parted executable..
/sbin/parted
[07-06-2016 16:20:50] ./icub_iso2usb.sh started: please do not unplug device sdb until finished.
Warning: more than 1024 cylinders (5128).
Not all BIOSes will be able to boot this device.
Warning: You requested a partition from 5128MB to 11.2GB.                 
The closest location we can manage is 5377MB to 11.2GB.
Is this still acceptable to you?
Yes/No? yes
Information: You may need to update /etc/fstab.                           

mv: cannot stat ‘/tmp/tmp.XcEVftLJs2/isolinux’: No such file or directory
[07-06-2016 17:14:02] ./icub_iso2usb.sh ERROR : failed to copy syslinux files on target device
```

Any idea?

Tobias-Fischer, 2016-06-07:
Hi,
I think you follow old instructions. The new instructions are here: http://wiki.icub.org/wiki/The_Linux_on_the_pc104#Burn_the_Debian_image_version_6.x_.28latest.29

jeljaik, 2016-06-08:
Hello @Tobias-Fischer I think you're right, for some reason I was following some old instructions maybe. So, I followed those instructions and did:

- [X] Download the latest image from the above link
- [X] Take USB drive at least of 8 GB size
- [X] Format the drive with fat32 filesystem
- [X] Unzip the content of the image archive into the root of the USB drive
- [ ] **Execute the "Make bootable" script as follows: Open a shell, then go to the folder `utils/linux` and execute the script `makeboot.sh` with root privileges (i.e. `sudo makeboot.sh`); then follow on-screen instructions**. 

The last instruction is giving me trouble since I can't execute the script. I tried `chmod +x makeboot.sh` or `chmod 777 makeboot.sh` but the permissions won't change :/ any idea?

pattacini, 2016-06-08:
@mbrunettini might give some help too

jeljaik, 2016-06-08:
Thanks @pattacini hope Matteo gives us a hand soon :)

lschilli, 2016-06-08:
It looks like the filesystem you are working on does not support permissions because it is fat32. So you cannot chmod something on that filesystem.
The permissions you are seeing are defined by the mount options when the fat32 volume is mounted (e.g. http://askubuntu.com/questions/96923/how-do-i-change-permissions-on-a-fat32-formatted-drive)
Often the default settings already include +x, but as it seems not in your case.

So you can solve it by changing how the drive is mounted. Another way is to not try to execute the script directly e.g. by ./makeboot.sh but by sh makeboot.sh (or sudo sh makeboot.sh). Replace sh with whatever interpreter the script requires (check the first line of the script). If this works as expected, maybe someone can add that to the manual.

mbrunettini, 2016-06-09:
You can simply execute by
bash  makeboot.sh [options]

mbrunettini, 2016-06-09:
Instructions updated.

rlober, 2016-06-09:
@mbrunettini @lschilli @jeljaik 

Thanks for the help guys! So recap. We tried `sudo sh makeboot.sh` and it did not work. This morning I tried `sudo bash makeboot.sh` per @mbrunettini  and it did work :+1:. This is odd because our shell should default to bash:

```
ps -p $$
 PID TTY          TIME CMD
 4948 pts/30   00:00:00 bash
```
but whatever...

Anyways, I just thought I would share my experience with the script.
First I ran `makeboot` without options giving:

```bash
icub@icub-desktop:/media/icub/C2F8-C2C6/utils/linux$ sudo bash makeboot.sh 
[sudo] password for icub: 
No destination partition was assigned!
makeboot.sh - To make the device bootable with syslinux
Usage: makeboot.sh [OPTION] partition_device
Options:
-b, --batch-mode 	batch, unattended mode. Use carefully! DANGEROUS!!!
-L, --LABEL STRING 	set device via LABEL if LABEL exists
-U, --UUID STRING	set device via UUID
Note:
  Device assignment priority: Partition name > UUID > LABEL (if not only one NAME-TYPE is assigned)
Ex:
To make "/dev/sdg1" bootable on GNU/Linux:
  makeboot.sh /dev/sdg1
To make device label "LABEL_STRING" bootable on GNU/Linux:
  makeboot.sh -L LABEL_STRING
To make device with UUID "UUID_STRING" bootable on GNU/Linux:
  makeboot.sh -U UUID_STRING
```

Realized it needed a drive so I ran:

```bash
icub@icub-desktop:/media/icub/C2F8-C2C6/utils/linux$ mount -l | grep dev/
/dev/sda4 on / type ext4 (rw,errors=remount-ro)
devpts on /dev/pts type devpts (rw,noexec,nosuid,gid=5,mode=0620)
/dev/sda1 on /boot/efi type vfat (rw)
/dev/sda2 on /media/icub/7a27a43c-d769-44e1-a93e-334e4d443493 type ext4 (rw,nosuid,nodev,uhelper=udisks2)
/dev/sdb1 on /media/icub/C2F8-C2C6 type vfat (rw,rce)
```
`sdb1` is the one.

So then I re-executed with the drive:
```bash
icub@icub-desktop:/media/icub/C2F8-C2C6/utils/linux$ sudo bash makeboot.sh /dev/sdb1
This command will install MBR and syslinux bootloader on this machine
--------------------------------------------
Machine: V-P8H67E:
Model: Generic Flash Disk (scsi)
Disk /dev/sdb: 16.6GB
Sector size (logical/physical): 512B/512B
Partition Table: msdos

Number  Start   End     Size    Type     File system  Flags
 1      1049kB  16.6GB  16.6GB  primary  fat32

--------------------------------------------
Are you sure you want to continue?
[y/n] y
OK! Let's do it!
--------------------------------------------
File system of /dev/sdb1: vfat
--------------------------------------------
sdb1 is not marked as bootable! The partition table of /dev/sdb:
--------------------------------------------
V-P8H67E:
Model: Generic Flash Disk (scsi)
Disk /dev/sdb: 16.6GB
Sector size (logical/physical): 512B/512B
Partition Table: msdos

Number  Start   End     Size    Type     File system  Flags
 1      1049kB  16.6GB  16.6GB  primary  fat32

--------------------------------------------
Do you want to mark it as bootable ?
[y/n] y
OK! Let's do it!
Running: parted -s /dev/sdb set 1 boot on
--------------------------------------------
Do you want to install mbr on /dev/sdb on this machine "V-P8H67E" ?
[y/n] y
OK! Let's do it!
--------------------------------------------
Do you want to install the SYSLINUX bootloader on /dev/sdb1 on this machine "V-P8H67E" ?
[y/n] y
OK! Let's do it!
A filesystem supporting Unix file mode for syslinux is required. Copying syslinux from FAT to /tmp/...
‘/media/icub/C2F8-C2C6/utils/linux/syslinux’ -> ‘/tmp/syslinux_tmp.LPWcDP/syslinux’
Running: /tmp/syslinux_tmp.LPWcDP/syslinux -d syslinux -f -i /dev/sdb1 
done!
//NOTE// If your USB flash drive fails to boot (maybe buggy BIOS), try to use "syslinux -d syslinux -fs /dev/sdb1", i.e. running with "-fs".
```

So firstly - kudos on the script! Apart from the bash issue, it is very straightforward. 

two notes:
1. It would be good to add the possible `[options]` for `makeboot` to the wiki. i.e. the drive to make bootable.
2. It would be nice to have a little detail on what `mbr` and `SYSLINUX bootloader` do. I am no linux guru so this would be nice for a noob such as myself. (It would be good to put defaults in the script for these to guide the noobs)

I will test the image either today or tomorrow to make sure everything went well. Closing for now!

mbrunettini, 2016-06-09:
@rlober : I'm happy it worked. The script is a common script used by several LIVE linux (eg. Clonezilla). 
As regards instructions: I prefer not to go too into details in the wiki page, otherwise It can be more misleading then now: MBR and syslinux are common linux commands, so just google them or check here:
 http://manpages.ubuntu.com/manpages/wily/man8/install-mbr.8.html 
 http://manpages.ubuntu.com/manpages/wily/man1/syslinux.1.html

lschilli, 2016-06-09:
@rlober : side note concerning:

> This is odd because our shell should default to bash

`sh` does not call your default shell (and it does not interpret `#!/path/to/interpreter`, because that is done by the kernel's program loader when a script is executed like e.g. `./script.sh`). Furthermore, the `sh` implementation can be distribution specific, e.g. in ubuntu it symlinks to `dash` and not to `bash`. That's why it makes a difference calling `sh some-script.sh` or `bash some-script.sh` or `./some-script.sh`

rlober, 2016-06-09:
@lschilli good to know thanks for the info!

frank-foerster, 2016-10-05:
I just installed this image and when test-booting it in my virtual machine I see the following error during boot time (ignore the last two characters, they were produced by my screenshot keys being accidentally captured by the virtual machine):
[image]

Is this error message about not being able to mount /dev/sda ok, or did something go wrong?

mbrunettini, 2016-10-05:
First, the image is not ment to be executed on a virtual machine, but on the real iCub Hardware: the startup process rely on hardware which can't be emulated on a VM.
Anyway, how did you produced the image? It seems that something is missing in the live filesystem.

frank-foerster, 2016-10-05:
I followed the instructions for the new images and just executed the script. We're running the image on a virtual machine first in order to adapt some network settings for our local network setup here etc. That hasn't been a problem with the old image. 
But if this error will go away on the real robot, then everything is fine. 
Is it supposed to produce/mount the persistence partition?

mbrunettini, 2016-10-05:
Just to be sure: which version have you downloaded and which script have you executed?

frank-foerster, 2016-10-05:
The image is the second-to-last image, coming in the zip-file: `icub-live_6.0-16.04.05-rt-amd64.zip`. The script was `/utils/linux/makeboot.sh` and I think I executed it with sudo bash <script_name> /dev/<usb_dev>, but don't nail me down on the latter, as I tried a lot of things yesterday.

mbrunettini, 2016-10-05:
The persistence in the above version is only a file. I think something wrong has happend while creating the USB.
Please try again and please check the following:

1. Take an USB drive at least with 8 GB size
2. Format the drive with fat32 filesystem
3. Open a shell, then go to the folder utils/linux and execute the script makeboot.sh with root privileges (i.e. `sudo bash makeboot.sh [PARTITION DEVICE]` ); then please follow on-screen instructions or reported errors

Let us know

frank-foerster, 2016-10-05:
My USB stick is slightly smaller than 8GB, it's 7.3GB. So maybe that's what caused the problem. 
I'm planning to use the old image before 6.0 again for the time being to test some hardware change but will update thereafter at some point. If I run into the same problems again I will post them here.

closed 2020-12-15 by pattacini