robotology/community #118: Benefits of GitHub based Q&A system over mailing list

pattacini, 2016-06-12:
Just to give a look in retrospect, here are the main advantages we managed to gain by using GitHub as our Q&A system, as taken from https://www.godaddy.com/garage/webpro/development/cool-github-hack.

#### It's asynchronous
For questions that fall into the “important, but not urgent” box, this model works really well. An individual can post a public question to another, and the recipient can get to it in the queue when time permits. No need for subscribing: **everything is public**.

#### It enables arbitrary length of response
Although some questions can be answered in 140 characters, others require a lengthier and more thought-full (as opposed to thought-free) response. This works wonderfully for that.

#### It’s persistent, one-to-many and public
Both of the points above could be applied to email as well. However, engaging in Q&A using this mechanism enables the interaction to be **indexed by search engines** and enables it to be found by others in the future who might have a similar question.

#### It builds reputation and community
All of the Qs and As that gather engagement not only show up in the Q&A thread, but also show up in the **activity history of the individual participants**. This gives a pulse to their activity over time when they participate. GitHub also highlights the group of individuals who have participated in the topic thread, giving the conversation a human face.

alecive, 2016-06-13:
It doesn't clutter everyone's inbox :smile:

lornat75, 2016-06-15:
Another positive point is that now that we have routed most of the specific questions from users, we can use the mailing list for important, general public notifications.

drdanz, 2016-06-17:
We can easily redirect questions to the repositories they belong, or assign them to someone

pattacini, 2016-06-17:
@drdanz actually, assigning tasks is a big pro

drdanz, 2016-06-17:
We can also fix typos, formatting, etc

closed 2020-12-08 by pattacini