robotology/community #119: New Yarp and iCub releases available

pattacini, 2016-06-13:
Dear @robotology/all-robotology-developers 

We have lately released new software binaries for `yarp` and `icub-main`. You'll find them available from the following resources:

- [**`Yarp 2.3.66`**](https://github.com/robotology/yarp/releases/tag/v2.3.66)
- [**`iCub 1.4.0`**](https://github.com/robotology/icub-main/releases/tag/v1.4.0)

Importantly, we have also introduced **new policies** for developing code in `yarp`, in particular regarding **branching**. Therefore, we warmly suggest you to look carefully at these [**contributing guidelines**](https://github.com/robotology/yarp/blob/master/.github/CONTRIBUTING.md).

Also, our intention is to extend in the near future these new policies to code development in `icub-main`.

pattacini, 2016-06-13:
cc @mbrunettini @drdanz

traversaro, 2016-06-13:
Cool innovation of this release: Windows binaries for `icub-main` and its dependencies are available for 64-bit compilers!

giodegas, 2016-06-13:
Just updated/re-built respective docker images on [dockerhub.com](http://dockerhub.com):
* giodegas/yarp
* giodegas/icube

pattacini, 2016-06-14:
Thanks @giodegas for maintaining [**Docker Hub**](https://hub.docker.com) repositories!

Just a comment on that: actually, the name of the robot is **iCub**, not **iCube** :wink:

giodegas, 2016-06-14:
Thank you @pattacini and I apologize for the typo.
All fixed!

pattacini, 2016-06-14:
Never mind! Thanks for the fix.

closed 2020-12-12 by pattacini