robotology/community #121: Work in progress pull requests.

drdanz, 2016-06-17:
It's happening more and more frequently that people (myself included) open pull requests for work in progress branches, for different reasons (testing if something breaks some build, requesting early feedback, etc).
I think this is a good thing, but sometimes it is not easy to spot whether a pull request is a real one, or a work in progress.

I suggest to follow [this workflow in gitlab](http://docs.gitlab.com/ce/workflow/wip_merge_requests.html), i.e. add `WIP:` or `[WIP]` at the beginning of the title of a "work in progress pull request" and remove it when the pull request is ready to be merged.
Even if this feature is not supported by github (in gitlab the merge button is disabled for this kind of merge requests), it will avoid accidental merges.

What do you think?

pattacini, 2016-06-17:
I fully agree.
Nice suggestion.

barbalberto, 2016-06-17:
We could also define WIP as a label to make it more visible

drdanz, 2016-06-17:
@barbalberto We already have a "Status: In Progress" label, we can use that, I think...

pattacini, 2016-06-17:
We could put this info in Yarp here: [**`CONTRIBUTING.md`**](https://github.com/robotology/yarp/blob/master/.github/CONTRIBUTING.md)

vtikha, 2016-06-17:
Excellent!

closed 2020-12-08 by pattacini