robotology/community #122: IplImage* has memory leak, how to release memory ?

pattacini, 2016-06-24:
@Tariq-Abuhashim commented on [Fri Jun 24 2016](https://github.com/robotology/stereo-vision/issues/12)

IplImage* cvImageL  allocates the memory, and cvReleaseImage( &cvImageL ) is supposed to release it. Looks like it only deletes the pointer, but not the physical memory allocated.
This seems to be caused by cvLoadImage:
IplImage* cvImageL = cvLoadImage( left_img_file_name.c_str() );

==8172== LEAK SUMMARY:
==8172==    definitely lost: 0 bytes in 0 blocks
==8172==    indirectly lost: 0 bytes in 0 blocks
==8172==      possibly lost: 0 bytes in 0 blocks
==8172==    still reachable: 88 bytes in 3 blocks
==8172==         suppressed: 0 bytes in 0 blocks
==8172== Rerun with --leak-check=full to see details of leaked memory

Any ideas on how to empty the allocated memory ?

---

@pattacini commented on [Fri Jun 24 2016](https://github.com/robotology/stereo-vision/issues/12#issuecomment-228285172)

It seems unrelated to `stereo-vision`, although `stereo-vision` still makes use of old OpenCV code.
Moving to Q&A...

pattacini, 2016-06-24:
Hi @Tariq-Abuhashim 

`cvReleaseImage` should actually free up the memory, if I'm not mistaken.
However, I'd warmly recommend you to rely on the C++ layer of OpenCV, thus replacing the use of `*IplImage` with `Mat`.

Therefore, look up the [`imread`](http://docs.opencv.org/2.4/doc/tutorials/introduction/display_image/display_image.html) method, which should basically do what `cvLoadImage` did.

lornat75, 2016-06-29:
it seems strange that `cvReleaseImage()` does not free all the previously allocated memory (e.g. by `cvLoadImage( )`. This would imply a huge memory leak in opencv.

BTW: the way I have found most convenient to pass a YARP Image to OpenCV is by creating a Mat object and initialize it from a IplImage pointer, i.e.:

```
ImageOf<PixelRgb> image;
/* fill image content */
Mat image_cv((IplImage *)image.getIplImage());
/* now image_cv can be used in opencv*/
```

In my understanding this is quite efficient because `image_cv` shares storage allocated by `image`. Memory is therefore managed by YARP.

You just have to be careful that `image_cv` and `image` share their content.

is there a better method? @vtikha

gsaponaro, 2016-06-29:
Don't know about the leak, but just a small side comment on YARP + OpenCV. I basically use the same approach as @lornat75, with the addition of [cvarrToMat](http://docs.opencv.org/2.4/modules/core/doc/operations_on_arrays.html#cvarrtomat) to make the code portable between OpenCV 2 and 3:

        Mat image_cv;
        image_cv = cvarrToMat(static_cast<IplImage*>(image_yarp->getIplImage()));

lornat75, 2016-06-29:
...and of course C++ style casts... I am getting old :older_man:

closed 2020-12-08 by pattacini