robotology/community #124: Opening a schematic file (*.sch)

mulkkyul, 2016-07-14:
Hi all,

I have some problems with the MTB3_001 board (tactile sensor board) - (https://github.com/robotology/icub-support/issues/275), so I'm trying to get schematic of it. What I found on the svn repository is this one:
https://svn.robotology.eu/repos/iCubHardware-pub/trunk/electronics/public/MTB/Schematics/MTB3_REV1/

But I'm not able to open the schematic file (*.sch) . What kinds of software should I use? (on Ubuntu or Windows.)
Any tips would be appreciated. Thanks.

pattacini, 2016-07-14:
@maggia80 might help in this case, I guess.

maggia80, 2016-07-14:
There is an error in the filename, but the pdf of the schematic is in the same folder and it is the MTB3-002.pdf

closed 2020-12-15 by pattacini