robotology/community #125: Where to get icub 4.0 image?

mulkkyul, 2016-07-19:
Hi all,

I'm trying to burn an image for our iCub (iCubDaejeon01) and I have some questions.

1. I can't find the image in http://www.icub.org/download/pc104-images/. Where can I get it?
2. Is it okay to follow this instruction? (http://wiki.icub.org/wiki/The_Linux_on_the_pc104_oldrepo)
 

FYI, <i>/VERSION</i> in the current USB is
```
Debian version 4.0

This is the iCub Debian Linux system,
Built on: Mon, 20 Dec 2013 16:00:39 +0200

For more info see http://icub.org

```

pattacini, 2016-07-20:
@mbrunettini is likely to be the one to give heads-up...

mbrunettini, 2016-07-20:
@mulkkyul : why do you need version 4.0 ? It's outdated

mulkkyul, 2016-07-20:
Hi @mbrunettini ,

I'm having a problem with booting the PC104 (https://github.com/robotology/icub-support/issues/278), so I wanted to give it a try with a fresh USB. I was kinda afraid that upgrading the iCub image can make it hard to detect our current PC104 problem. That's why I was looking for the exact same image (4.0) rather than the latest one :)

mbrunettini, 2016-07-20:
Please try the version 6 (the latest one), first.
If it doesn't work, I will find a way to provide you the ver 4.0
Please remember that :
1) reading your past issues, it seems that you have power issues, not related to the System Image
2) during power issues, you can occur in filesystem or USB memory hardware corruptions
So, please use a new USB memory (of at least 8 GB) and Let me know

mulkkyul, 2016-07-20:
Thanks! I'll try with the latest one and let you know.

mulkkyul, 2016-07-21:
Hi @mbrunettini ,
I tested with the latest one (6.0) and I still have the PC104 booting problem. As you said, it might be power issues or others. Once I solve that PC104 issue, I'll update the robot with the latest image, so image 4.0 won't be necessary. Thanks!

closed 2020-12-15 by pattacini