robotology/community #126: Several issues with blades for iCubParis02

jeljaik, 2016-07-22:
Hello guys,

After solving many issues we had with the hardware of our two iCubs, I was trying today to setup the red ball demo, however I found the blades status quite unstable. I might need a bit of assistance on different issues that I will try to keep in this single issue. Let me start with the following problem: 

When I do `ssh icub-b1` this is what I get:
```shell
ssh icub-b1
Linux icub-netboot 3.2.0-4-amd64 #1 SMP Debian 3.2.51-1 x86_64

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Exporting iCub variables for: iCubParis02
```
After which the terminal shows: 
`icub@no` instead of the expected `icub@icub-b1`

If I try to do `sudo hostname icub-b1` this is what I get:
```shell
icub@no:~$ sudo hostname icub-b1
sudo: unable to resolve host icub-b1
```
I have the same problem with all three blades. Can you help me diagnose this and the other issues I will mention after this one is solved?

mbrunettini, 2016-07-26:
On one blade, please do the following commands:
`cat /etc/issue`
`cat /etc/rc.local`
And please post here the results

jeljaik, 2016-07-26:
Right away!

```
cat /etc/issue
Debian GNU/Linux stretch/sid \n \l
```

```
cat /etc/rc.local
#!/bin/sh -e
#
# rc.local
#
# This script is executed at the end of each multiuser run level
# Make sure that the script will "exit 0" on success or any other
# value on error.
#
# In order to enable or disable this script just change the execution
# bits.
#
# By default this script does nothing.

/root/set-hostname-from-net.sh
mount -a -t nfs

/root/force_ntp_update.sh
exit 0
```

mbrunettini, 2016-07-26:
Thanks, can you please issue also
`cat /root/set-hostname-from-net.sh`

jeljaik, 2016-07-26:
```
#! /bin/sh
​_HOSTNAME_​=""
LEASES_FILE_PATH="/var/lib/dhcp"

get_hostname_from_dns() {
 if [ "$1" != "" ]
 then
    IP_ADDR=$( ip addr show $1 | grep inet | grep $1 | awk '{ print $2 }' | sed -e 's/\/24//')
    if [ "$IP_ADDR" != "" ]
    then
        ​_HOSTNAME_​=$( host $IP_ADDR |  awk '{ print $5 }' | cut -d'.' -f1 )
    fi
 fi
}

get_hostname_from_leases() {
 if [ "$1" != "" ]
 then
   if [ -f "${LEASES_FILE_PATH}/dhclient.${1}.leases" ]
   then
     ​_HOSTNAME_​=$( grep "host-name" ${LEASES_FILE_PATH}/dhclient.${1}.leases | awk '{ print $3 }' | sed -e 's/"//g' -e 's/;//g' )
   fi 
 fi
}

get_hostname_from_dns eth0
if [ "${​_HOSTNAME_​}" = "" ]
then
    get_hostname_from_leases eth0
fi
if [ "${​_HOSTNAME_​}" = "" ]
then
    get_hostname_from_dns eth1
fi
if [ "${​_HOSTNAME_​}" = "" ]
then
    get_hostname_from_leases eth1
fi
if [ "${​_HOSTNAME_​}" = "" ]
then
    exit 1
fi
echo $​_HOSTNAME_​
hostname "$​_HOSTNAME_​"
exit $?
```

mbrunettini, 2016-07-26:
It looks like a strange mix.. 
A script created for the Wheezy release on Debian etch (older).. really odd..
can you try to 
`sudo /root/set-hostname-from-net.sh`

jeljaik, 2016-07-26:
For example on `icub-b1` I get
```
sudo: unable to resolve host no
[sudo] password for icub:
no
```

mbrunettini, 2016-07-26:
ok, then:
`su -
/root/set-hostname-from-net.sh`

jeljaik, 2016-07-26:
```
icub@no:~$ sudo su
sudo: unable to resolve host no
bash: warning: setlocale: LC_ALL: cannot change locale (fr_FR.UTF-8)
root@no:/home/icub# /root/set-hostname-from-net.sh 
no
```

mbrunettini, 2016-07-26:
NO sudo
`su -`

jeljaik, 2016-07-26:
Oops, sorry

```
icub@no su - /root/set-hostname-from-net.sh
No passwd entry for user '/root/set-hostname-from-net.sh
```

mbrunettini, 2016-07-26:
Please read carefully, thanks:
`su - `
`/root/set-hostname-from-net.sh`
**Usually, 2 LINES = 2 SEPARATE COMMANDS**

jeljaik, 2016-07-26:
Relax, no need for captions ;) I was having some issue using just `su`

Here you go,
```
icub@no:~$ su - 
Password: 
-su: warning: setlocale: LC_ALL: cannot change locale (fr_FR.UTF-8)
-su: warning: setlocale: LC_ALL: cannot change locale (fr_FR.UTF-8)
root@no:~# /root/set-hostname-from-net.sh 
no
```

jeljaik, 2016-07-28:
ping!  :bell::bow:

mbrunettini, 2016-07-29:
Cut'n'paste the following command and execute on icub-b1
`host 10.0.0.51`
and copy here the result.
Thanks

jeljaik, 2016-07-29:
Sure, thanks

```
icub@icub-desktop:~$ ssh icub-b1
Linux icub-netboot 3.2.0-4-amd64 #1 SMP Debian 3.2.51-1 x86_64

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
-bash: warning: setlocale: LC_ALL: cannot change locale (fr_FR.UTF-8)
-bash: warning: setlocale: LC_ALL: cannot change locale (fr_FR.UTF-8)
Exporting iCub variables for: iCubParis02
-bash: warning: setlocale: LC_ALL: cannot change locale (fr_FR.UTF-8)
icub@no:~$ host 10.0.0.51
;; connection timed out; no servers could be reached```

mbrunettini, 2016-07-29:
Again on icub-b1:
`cat /etc/resolv.conf`
?

jeljaik, 2016-07-29:
```bash
icub@no:~$ cat /etc/resolv.conf
cat: /etc/resolv.conf: No such file or directory
```

jeljaik, 2016-07-29:
I also tried to see if he file is there, but that is what I get

```bash
icub@no:/etc$ ls | grep  resolv
resolv.conf
resolvconf

icub@no:/etc$ cat resolv.conf
cat: resolv.conf: No such file or directory
```

mbrunettini, 2016-07-29:
on icub-srv :
`ls -la /srv/tftp`
`ls -la /srv/tftp/wheezy-x86_64`
`cat cat /srv/tftp/wheezy-x86_64/etc/issue`
`cat cat /srv/tftp/wheezy-x86_64/etc/resolv.conf`

jeljaik, 2016-07-29:
```bash
icub@icub-srv:~$ ls -la /srv/tftp
total 20
drwxr-xr-x  5 tftp tftp 4096 Oct  1  2014 .
drwxr-xr-x  4 root root 4096 Sep 24  2014 ..
drwxr-xr-x  3 tftp tftp 4096 Sep 25  2014 tftpboot
drwxr-xr-x 22 root root 4096 Jul 22 16:17 wheezy-x86_64
drwxr-xr-x 24 root root 4096 Sep 24  2014 wheezy-x86_64.old
```

```
icub@icub-srv:~$ ls -la /srv/tftp/wheezy-x86_64
total 104
drwxr-xr-x  22 root root  4096 Jul 22 16:17 .
drwxr-xr-x   5 tftp tftp  4096 Oct  1  2014 ..
drwxr-xr-x   2 root root  4096 Jul 22 16:18 bin
drwxr-xr-x   2 root root  4096 Jul 22 11:17 boot
drwx------   2 root root  4096 Jul 22 15:26 .cache
drwxr-xr-x   3 root root  4096 Dec 11  2013 dev
drwxr-xr-x 136 root root 12288 Jul 26 16:51 etc
drwxr-xr-x   3 root root  4096 Dec 11  2013 home
lrwxrwxrwx   1 root root    30 Dec 12  2013 initrd.img -> /boot/initrd.img-3.2.0-4-amd64
drwxr-xr-x  18 root root  4096 Jul 22 16:17 lib
drwxr-xr-x   2 root root  4096 Jul 21 18:31 lib64
drwxr-xr-x   2 root root  4096 Oct 27  2015 media
drwxr-xr-x   2 root root  4096 Jan  5  2016 mnt
drwxr-xr-x   3 root root  4096 Dec 16  2013 opt
drwxr-xr-x   2 root root  4096 Sep 23  2013 proc
drwx------  10 root root  4096 Jul 26 15:31 root
drwxr-xr-x   7 root root  4096 Dec 11  2013 run
drwxr-xr-x   2 root root 12288 Jul 22 14:55 sbin
drwxr-xr-x   2 root root  4096 Dec 11  2013 srv
drwxr-xr-x   2 root root  4096 Jul 14  2013 sys
drwxrwxrwt   2 root root  4096 Jul 26 15:19 tmp
drwxr-xr-x  10 root root  4096 Jul 22 16:17 usr
drwxr-xr-x  12 root root  4096 Jul 26 15:19 var
lrwxrwxrwx   1 root root    26 Dec 12  2013 vmlinuz -> boot/vmlinuz-3.2.0-4-amd64
```

```
icub@icub-srv:~$ cat /srv/tftp/wheezy-x86_64/etc/issue
Debian GNU/Linux stretch/sid \n \l
```

```
icub@icub-srv:~$ cat /srv/tftp/wheezy-x86_64/etc/resolv.conf
cat: /srv/tftp/wheezy-x86_64/etc/resolv.conf: No such file or directory
```

mbrunettini, 2016-07-29:
Why there is a 
`wheezy-x86_64.old`
and the 
`wheezy-x86_64` 
has been written on 22 July?
What has been changed and why? It seems it is in a wrong state.

jeljaik, 2016-07-29:
Yes, 
Unfortunately as I stated at the beginning, this is just the state in which I found the blades. Maybe @ahoarau or @rlober have some idea.

mbrunettini, 2016-07-29:
Then, please restore the system to its original state and let me know

jeljaik, 2016-07-29:
Would you be so kind to tell me what's the best way to do that please?

mbrunettini, 2016-07-29:
If I don't know what has been changed since the system has been installed, of course I am not able to do it.

jeljaik, 2016-07-29:
Ok, so, since neither do I know what was changed in the system, is there any way I can just reset the blades to "factory settings" ?

mbrunettini, 2016-07-29:
You can:

1. Try to delete the directory `wheezy-x86_64` and rename `wheezy-x86_64.old' to `wheezy-x86_64`. And hope it works (of course we cannot be sure if other things has been changed)
2. write a clean version of wheezy-x86_64 (we can provide it, but it's 6 GB). this is not resolutive, too
3. reinstall the whole icub-srv from the vmware image (120 GB)

jeljaik, 2016-07-29:
Thanks @mbrunettini I will analyse these options and get back to you. 

Thanks for your time.

lornat75, 2016-08-02:
Hi @jeljaik, 
looking at the solutions suggested by Matteo looks like they are in order of complexity. I would try 1), then 2) then 3). Please let us know how things go we can help in the steps.

jeljaik, 2016-08-02:
Thanks @lornat75 for the suggestion. 
So far I have tried option 1 and rebooted `icub-src` but then I can’t `ssh icub-bx` getting the following message:

```
icub@icub-srv:~$ ssh icub-b1
ssh_exchange_identification: Connection closed by remote host
```

lornat75, 2016-08-02:
Assuming the backup copy of `wheezy-x86_64` (`wheezy-x86_64.old`) was taken before changing the system, I fear solution 2) would not do much different. Let's see if @mbrunettini has some suggestions on  how to fix the problem you have with ssh.

jeljaik, 2016-08-02:
@lornat75 True, I'll wait for his reply.

mbrunettini, 2016-08-02:
`ssh_exchange_identification: Connection closed by remote host`
I must say that I've never seen this error before.
The netboot system (at least, we have to check the whole icub-srv later) seems to be really compromised: I suggest to delete and recreate it starting from our images.
We can provide a working version of 
`/srv/tftp/wheezy-x86_64 `
to replace the failing one. It is a compressed file of about 6GB, I can provide a download link 
how do you think?

jeljaik, 2016-08-02:
@mbrunettini I can try the new image, yes. I'll be waiting for the link. Thank you so much.

mbrunettini, 2016-08-02:
I'm uploading the image to the cloud. It's a .tar.gz of 3.1GB (I removed not used stuff).
I will post the link here when it's available
Once you have downloaded it, please:

1.  switch off the 3 blades  
2.  check on icub-srv if there is enough space on /srv/tftp to keep the two old versions of
`/srv/tftp/wheezy-x86_64` (at least 10 GB will be necessary)
3.  rename the two old version of 
`/srv/tftp/wheezy-x86_64`
4.  copy the downloaded archive to icub-srv in the following path
`/srv/tftp`
5.  extract it

jeljaik, 2016-08-02:
I will do @mbrunettini thanks!

mbrunettini, 2016-08-02:
You can download it here:
https://istitutoitalianotecnologia-my.sharepoint.com/personal/matteo_brunettini_iit_it/_layouts/15/guestaccess.aspx?guestaccesstoken=iPdeNt3TVUszkR%2bhB18wve8vA6Htw7s%2fwV19eP9AEj4%3d&docid=0ae02626af7a34598b60509bc54adfa02&rev=1
Let us know

jeljaik, 2016-08-02:
Alright, so a few things have changed! which I guess is something good. 

I followed the steps you listed above and after rebooting `icub-srv` and the blades, when I `ssh` into any of them I notice the following things: 

1. The password is changed to `icubAdmin` (ours was `icub` for all of them) which I assume is the default and a normal behaviour.

2. Regardless of where I `ssh` they all say `icub-b2`. Maybe this is normal too and an additional configuration step is needed. If so, please let me know. 
[image]

3. As per comment https://github.com/robotology/QA/issues/126#issuecomment-236125284 I get the following from `icub-b1` which is also good, since before it was always empty. 
```
icub@icub-b2:~$ cat /etc/resolv.conf 
domain icub.local
search icub.local
nameserver 10.0.0.1
```

mbrunettini, 2016-08-03:
Good! Well, not perfect, but a big step further.
Please do the following checks:

- [ ] `ip addr`
- [ ] `nslookup icub-b1`
- [ ] `mount`

Thanks

jeljaik, 2016-08-03:
Always on `icub-b1` (hoping I assumed correctly :sweat_smile:)

```
icub@icub-b2:~$ ip addr
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 16436 qdisc noqueue state UNKNOWN
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
    inet6 ::1/128 scope host
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP qlen 1000
    link/ether 84:34:97:11:b0:ec brd ff:ff:ff:ff:ff:ff
    inet 10.0.0.51/24 brd 10.0.0.255 scope global eth0
    inet6 fe80::8634:97ff:fe11:b0ec/64 scope link
       valid_lft forever preferred_lft forever
3: eth14: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc mq state DOWN qlen 1000
    link/ether 84:34:97:11:b0:ed brd ff:ff:ff:ff:ff:ff
```

```
icub@icub-b2:~$ nslookup icub-b1
Server:		10.0.0.1
Address:	10.0.0.1#53

Name:	icub-b1.icub.local
Address: 10.0.0.51
```

```
icub@icub-b2:~$ mount
sysfs on /sys type sysfs (rw,nosuid,nodev,noexec,relatime)
proc on /proc type proc (rw,nosuid,nodev,noexec,relatime)
udev on /dev type devtmpfs (rw,relatime,size=10240k,nr_inodes=4104691,mode=755)
devpts on /dev/pts type devpts (rw,nosuid,noexec,relatime,gid=5,mode=620,ptmxmode=000)
tmpfs on /run type tmpfs (rw,nosuid,noexec,relatime,size=3284936k,mode=755)
10.0.0.1:/pxeroot on / type nfs (rw,relatime,vers=3,rsize=262144,wsize=262144,namlen=255,hard,nolock,proto=tcp,port=2049,timeo=7,retrans=3,sec=sys,local_lock=all,addr=10.0.0.1)
tmpfs on /run/lock type tmpfs (rw,nosuid,nodev,noexec,relatime,size=5120k)
tmpfs on /run/shm type tmpfs (rw,nosuid,nodev,noexec,relatime,size=6569860k)
none on /tmp type tmpfs (rw,relatime)
none on /run type tmpfs (rw,relatime)
none on /var/tmp type tmpfs (rw,relatime)
none on /var/log type tmpfs (rw,relatime)
none on /var/mail type tmpfs (rw,relatime)
none on /var/spool/cron type tmpfs (rw,relatime)
tmpfs on /run/lock type tmpfs (rw,nosuid,nodev,noexec,relatime,size=5120k)
tmpfs on /run/shm type tmpfs (rw,nosuid,nodev,noexec,relatime,size=6569860k)
rpc_pipefs on /var/lib/nfs/rpc_pipefs type rpc_pipefs (rw,relatime)
binfmt_misc on /proc/sys/fs/binfmt_misc type binfmt_misc (rw,nosuid,nodev,noexec,relatime)
10.0.0.1:/exports/code on /usr/local/src/robot type nfs (rw,relatime,vers=3,rsize=262144,wsize=262144,namlen=255,hard,proto=tcp,timeo=600,retrans=2,sec=sys,mountaddr=10.0.0.1,mountvers=3,mountport=60197,mountproto=udp,local_lock=none,addr=10.0.0.1,_netdev)
10.0.0.1:/exports/local_yarp on /home/icub/.local/share/yarp type nfs (rw,relatime,vers=3,rsize=262144,wsize=262144,namlen=255,hard,proto=tcp,timeo=600,retrans=2,sec=sys,mountaddr=10.0.0.1,mountvers=3,mountport=60197,mountproto=udp,local_lock=none,addr=10.0.0.1,_netdev)
```

mbrunettini, 2016-08-03:
Everithing seems ok.
To fix the hostname issue, on icub-b1 you have to modify the following file 
`/root/set-hostname-from-net.sh`
as follows: on line 27 change from
`get_hostname_from_dns`
to
`get_hostname_from_dns eth0`
the reboot the 3 blades.

jeljaik, 2016-08-03:
Perfetto!
I can now `ssh` into each one of them and see the proper host names. Overall, the cluster seems "stable" now.  I will proceed upgrading source codes and so on. Should I have any additional problem I will open a different issue. Maybe this one can be renamed to the  main problem that was tackled ... **Restoring Unstable Cluster Setup** (?)
Thanks for your patience.

mbrunettini, 2016-08-03:
Issue renamed as requested.
@jeljaik ,please remember to change the icub password on the blades (changing on one will also replicate on the others) to the original one.
Can we close this issue?

jeljaik, 2016-08-03:
Sure I will change it. 
Thanks. Closing!

closed 2020-12-15 by pattacini