robotology/community #127: Reading yarp ports

KevinAStein, 2016-07-26:
Hi,
I want to read force-torque and joint sensors from iCub to estimate ZMP and COM position and velocity using wholeBodyDynamicsTree. Since the readings are quite noisy I tend to use mean values. The results are okay reading every port approximately 50 times. This takes quite long (~200ms).

Is there a way to parallelize this? In what rate is the sensor data read and streamed?

thanks,
Kevin

@hu-yue

apaikan, 2016-07-26:
@KevinAStein  What do you exactly mean with paralleling it? 

Anyway a quick way to see the port data streaming frequency is using yarp read using `envelop` and see the time-stamp differences as follows: 

```
$ yarp read ... /portname envelope 
```

traversaro, 2016-07-26:
I @KevinAStein , external forces in `wholeBodyDynamicsTree` is published on the ports every period of wholeBodyDynamicsTree, that is by default set to `10 ms`. You can modify it with the `rate` period (sorry for the misleading name, I need to fix it). However decreasing the period you would probably get the same estimates streamed for multiple times, as most of the sensors of the robot publish data with a period of ~10 ms. 
However I suggest that you use the low-pass filters embedded in `wholeBodyDynamicsTree`  (see how to set their parameters with `wholeBodyDynamicsTree --help`) to reject noise in the estimation, or you use some "classical" low pass filter on your side, and you probably will need less then 50 samples to get a meaningful estimate. : )

pattacini, 2016-07-29:
@KevinAStein have the recommendations given above been useful for you?

KevinAStein, 2016-07-29:
thanks for your help. Using the already implemented filter made things a lot easier.

closed 2020-12-08 by pattacini