robotology/community #128: Qt 5.7 & Yarp failed link on Windows 10 (also iCub)

pattacini, 2016-07-27:
@gregoire-pointeau commented on [Tue Jun 21 2016](https://github.com/robotology/yarp/issues/808)

Hello,
My installation was working on windows 10 until an automatic update last Friday (17 June).
Since I have the following problem when I launch any GUI (`yarpmanager`, `iCubGui`...)
I have the following error message: 
for `yarpmanager`:
> `?setSelectionModel@QListWidget@@UAEXPAVQItemSelectionModel@@@Z process entry point not found in the library link library C:\Robot\yarp\build\bin\Release\yarpmanager.exe`

for `iCubGui`:
> `?getProcAddress@QOpenGLContext@@QBEP6AXXZPBD@Z process entry point not found in the library link library C:\Robot\robotology\Qt\5.7\msvc2013\bin\Qt5OpenGL.dll`

I reinstalled Qt, re-pull everything, and rebuilt several times everything.

My configuration:
- Windows 10
- MVS 12 2013
- Cmake 3.5.2
- Qt 5.7

`Qt5_DIR: C:\Robot\robotology\Qt\5.7\msvc2013\lib\cmake`

INCLUDE has:
`C:\Robot\robotology\Qt\5.7\msvc2013\include`

PATH has:
`C:\Robot\robotology\Qt\5.7\msvc2013\lib`
`C:\Robot\robotology\Qt\5.7\msvc2013\bin`

Does anyone had a similar problem using windows 10?

Thanks

---

@gregoire-pointeau commented on [Tue Jul 26 2016](https://github.com/robotology/yarp/issues/808#issuecomment-235257339)

Any update on it anyone ?

---

@drdanz commented on [Tue Jul 26 2016](https://github.com/robotology/yarp/issues/808#issuecomment-235403730)

@gregoire-pointeau I'm sorry, I'm not a Windows user, perhaps @randaz81, @pattacini or @mbrunettini saw something similar? Are you sure that you don't have more than one qt5 installation in your path? I've seen strange behaviours on windows with recent versions of CMake that include qt5 dlls in its path.
Anyway it looks to me something related to your setup, not a bug in yarp, therefore I'm closing this, please reopen it if you find out that the bug is actually in YARP, or open a new one in the robotology/QA if you need more support with this issue.

pattacini, 2016-07-27:
@gregoire-pointeau @drdanz I've never experienced such a problem.

I've got a fairly different env status though:

```
INCLUDE=C:\dev\Qt-5.4.1_v12_x86\include
LIB=C:\dev\Qt-5.4.1_v12_x86\lib
PATH=C:\dev\Qt-5.4.1_v12_x86\bin
Qt5_DIR=C:\dev\Qt-5.4.1_v12_x86\lib\cmake\Qt5
```

Very different actually...

pattacini, 2016-07-27:
@gregoire-pointeau did you compile `Qt` manually or rely on the package installed alongside robotology components?

mbrunettini, 2016-07-27:
As far as I see, it was installed manually (by sources or by QT binaries, NOT our packages), we haven't QT version 5.7 in our packages..

gregoire-pointeau, 2016-07-27:
Indeed I installed by hand Qt 5.7 before.

Now I've downgraded to the one automatically installed from the packages and I have the same configuration of env than Ugo. And this ime my error is:

`This application failed to start because it could not fin or loar the Qt platform plugin "windows" in "". Reinstalling the application may fix this problem.`

I reinstalled everything several times already.

This happens each time I try to launch a graphical application (example: `yarplogger`).

vtikha, 2016-07-27:
@gregoire-pointeau this seems to be related to the failure of finding the correct dlls/plugins etc..is qt linked properly in your environmental variables?

gregoire-pointeau, 2016-07-27:
Yes I have the same env than Ugo.
my latest installation is the one from the yarp binary.

drdanz, 2016-07-27:
@gregoire-pointeau That was exactly the error I saw when CMake was in the path... please check your `PATH` variable and ensure that CMake (if it is there) is at the end.
Otherwise check if any of the folders in `PATH` contain the Qt5 dlls

gregoire-pointeau, 2016-07-27:
I have nothing related to CMAKE in my paths (neither system than user).
The console app run perfectly (yarp server for example).
Should I had it ?

pattacini, 2016-07-29:
As side note, I usually don't put CMake in the path

lornat75, 2016-08-02:
I had the same issue, in the end due to CMake being in the path (btw: it is useful to have CMake in the path for command line usage :racehorse: ). However the problem happens if you have mismatching versions of qt libraries in the path, one way to find out is to wipe your path and add to it incrementally.

lornat75, 2016-08-02:
As @drdanz suggested, check if you don't have conflicting versions of qt5 dlls in your path. One way to do is to wipe your path and incrementally add to it.

drdanz, 2016-08-02:
@gregoire-pointeau There are 2 parts of the `PATH` environment variable, one is the system path, and one is the user path, I'm not sure In which order they are used, but if you have qt path on one path and cmake path in the other, cmake might be considered before even if it is the last one in its string.

drdanz, 2016-08-02:
Also search "rapid environment editor" for a very useful tool to use instead of the defaul editor for environment variables

gregoire-pointeau, 2016-08-16:
@drdanz REE helped me a lot thanks.

I've removed Cmake completely from my path, and keep Qt of course.

But now I have a different behavior. When I launch the release of `yarplogger` I get: `This application failed to stat because it could not find or load the Qt platform plugin "windows" in "".`

But if I compile in debug, and launch the `yarplogger-qt` in debug, everything works perfectly.

drdanz, 2016-08-16:
@gregoire-pointeau I've never seen this error, sorry... Anyway, doing a quick search on google returned many results, see for example if one of these can help:

* http://stackoverflow.com/questions/20495620/qt-5-1-1-application-failed-to-start-because-platform-plugin-windows-is-missi
* http://stackoverflow.com/questions/16773789/qt5-static-build-yields-failed-to-load-platform-plugin-windows/17271172#17271172
* http://stackoverflow.com/questions/21268558/application-failed-to-start-because-it-could-not-find-or-load-the-qt-platform-pl

pattacini, 2016-08-25:
Any progress on this?

gregoire-pointeau, 2016-08-25:
Well, things are that I don't (and can't for a long term period) add some stuff in the `iCub` and `yarp` code. So for know, the application that need Qt (the GUI) I launch the debug exe, and it works.
It's only on an external computer not linked to the robot.

gregoire-pointeau, 2016-08-25:
Some solutions needs to change some call to lib in the cmake or in the projects, but I don't want to start writing in the cmake of yarp or icub.
Thansk anyway.

closed 2020-12-15 by pattacini