robotology/community #129: Yarp crahes when connecting to ports

KevinAStein, 2016-07-29:
Hi,
I have a program using yarp ports as described in http://www.yarp.it/rpc_ports.html
It used to work fine since lately we started using the ```devel``` branch of yarp because @traversaro developed a floating base estimation that requires that branch. Now every time I try to connect to a RPC or buffered port it results in a segmentation fault. Neither ```yarp rpc /portname``` nor ```yarp read ... /portname``` work.
 Can somebody help me with this?
Thanks

 @hu-yue

traversaro, 2016-07-29:
There have been recently some fixes related to memory access in the `devel` branch, are you using the latest commit in the `devel` branch?

traversaro, 2016-07-29:
By the description of the problem, it could be something related to having two different version of YARP in the system. Can you check that you have (for example) a unique `libYARP_sig.so` (or at a maximum two, one for the `build` folder and one for the install folder)?

pattacini, 2016-07-29:
This is a yarp related issue. Moving to yarp repository

pattacini, 2016-07-29:
Issue moved to [robotology/yarp #836](https://github.com/robotology/yarp/issues/836) via [**ZenHub**](https://www.zenhub.com/)

closed 2020-12-15 by pattacini