robotology/community #131: Compliance gestion with goToPose order

misaki43, 2016-08-10:
Hello! 

In my program I need the robot to be compliant, except at one moment. At this special moment, the robot has to go back to the initial position.
To put the robot in the compliant mode, I use the following code for each joints:
`iimp->setImpedance(j,leftArmJointsStiffness[j],leftArmJointsDamping[j]);
 ictrl->setTorqueMode(j); 
 itrq->setRefTorque(j, 0.0); // should be totally compliant ? How is "setImpedance" taking into account after that ?
imode->setInteractionMode(j,VOCAB_IM_COMPLIANT);
`
Then, to put the robot back to the "home position" I use the code:
`client.goToPose(xinit,od); `
(that is a iCartesianControle).

After that, I use again the following code to put the robot in the compliant mode:
`ictrl->setTorqueMode(j); 
 itrq->setRefTorque(j, 0.0);   
// should it be totally compliant? How is the "setImpedance" taking into account?   
imode->setInteractionMode(j,VOCAB_IM_COMPLIANT);`

The issue is that the second time I ask it to be compliant in the torque mode, it stays on the position direct mode, I don't know why.
I tried to understand and I have seen that if I use the commands:
`ictrl->getControlMode(j, &control_mode);
 imode->getInteractionMode(j, &interaction_mode);
`
So the problem is that after the "goToPose" order, the robot is still in the position direct mode, according to the "ictrl" (IControlMode2). The Controle Mode seems to be not update.

Can you explain why and what I am supposed to do?

Moreover, I tried to understand better the control mode of the robot. I wanted to looked the "goToPose()" function, but I didn't find it (I only find the CartesianControl.h file but not the cpp function linked). Could you help me please? 

Finally, I would like to verify something. I am only interested on the position control of the robot (goToPose and getPose) and by the possibility to change the compliance of the robot. Have I a cleaner way to put the robot in the compliance mode, without passing from torque control to position control?

Thanks!

pattacini, 2016-08-10:
Hi @misaki43 

I'm afraid there's a bit of confusion between `torque` and `compliant-mode`: the former is one of the available **control mode**, whereas the latter is one of the two allowed **interaction mode**.

While `position-direct` and `torque` take a mutually exclusive control of the joints realizing two utterly different tasks, `position-direct` can be used either in `stiff` or `compliant` mode, which are in turn the two interaction modes we provide. On the other hand, `torque` does not go along with any specific interaction mode, given that it takes over the low level control outright.

To summarize:

**CM**=Control Mode; **IM**=Interaction Mode

| Joint in CM | IM = Stiff | IM = Compliant |
| ------------- | ------------- | ------------- |
| **Position-Direct**  | **Rigidly** controlled toward the target | **Spring effect** while controlled toward the target |
| **Torque**  | The torque reference is directly assigned and tracked (e.g. zero-g) | The torque reference is directly assigned and tracked (e.g. zero-g) |

It comes out that when the joint is in `torque`, the control is not affected by **IM**.

I don't know exactly where I can point you at for finding out more on this topic. Perhaps @randaz81 and/or @traversaro can give better hints on what I've reported above and - more importantly - on where the documentation is.

Coming to your specific problem, the correct instructions to put a joint in compliant mode is (let's assume you're already in `position-direct`):
```cpp
IInteractionMode  *imode;
IImpedanceControl *iimp;

driver->view(imode);
driver->view(iimp);

iimp->setImpedance(j,stiffness_value,damping_value);
imode->setInteractionMode(j,VOCAB_IM_COMPLIANT);
```

At this point, it's not necessary to disable compliance when you want to drive the robot back to its home position, unless you want to do so for a specific reason (maybe for precision? but it's a mere home position in the end). However, to do that:

```cpp
imode->setInteractionMode(j,VOCAB_IM_STIFF);
```

The Cartesian control is agnostic with respect to how its reference commands will be actually implemented by the firmware depending on the current **IM**. Therefore you can enable and disable the compliant mode at will with no particular care.

On the contrary, `torque` is a control mode, and as such it does interfere with `position-direct`. As a result, you have to be careful with the switching. If the movement yielded by a call to the `goToPose` service is still ongoing, then you cannot safely switch to `torque`. Well, you can try it out, but the Cartesian controller will take over again and switch it back to `position-direct`. This check is being performed continuously until the robot has come to a complete stop (i.e. has reached its final target or is asked to stop straightaway).

Thereby, you might consider calling the following methods before changing the control mode when you're steering the robot in the operational space: `checkMotionDone`, `waitMotionDone`, `stopControl`.

Hope this better clarifies the matter.

misaki43, 2016-08-11:
Hello, and thanks to help me again!
Your table is clear. I understand the global idea but some details are not totally clear for me: 
1. How is define the virtual "spring stiffness and damping" when we are in "Position-Direct" and "compliant mode"? Is it thanks to the `setImpedance` function? 
2. If I want the robot to follow my movement, I think about this code:
`client.getPose(x,o);
client.goToPose(x,o);`
The idea is to put the equilibrium position of the virtual spring to the actual position of the robot to maintain it in the more compliant way. Is it a correct way to allow the robot to follow a movement?
3. About the issue to change the control mode and interaction mode. When I use the commands:
`
	

client.setTrajectoryTime(3.0);
				client.setInTargetTol(0.1);
				client.goToPoseSync(xinit,od);
				client.waitMotionDone(0.04);

pattacini, 2016-08-11:
Hi @misaki43 

I'll try to answer your questions below.

1. From an excerpt of the (old by now) manual, we have:

  > Impedance control mode (now it is called **IM=COMPLIANT**) allows you to control the joint position and its compliance. In particular, you can control the equilibrium position of a **virtual spring** (using the standard `yarp::dev::IPositionControl` interface in the joint space or the `yarp::dev::ICartesianControl` interface in the operational space) and its **stiffness/damping** (using the `yarp::dev::IImpedanceControl` interface).
The control is implemented in the DSP firmware as follows:
  [image]
  > Firstly, a reference torque is computed, accordingly to the input position and the commanded stiffness/damping parameters (_Hooke_'s law). Secondly, the reference torque is tracked by a _PID_ controller (same gains used by the `torque` control mode) regulating the voltage _PWM_. By tuning the stiffness parameters, you can thus make the robot joint feel like a hard or soft spring, while maintaining control on the desired joint position.

  The stiffness and damping values of the virtual spring can be modified by means of the `setImpedance`  method as you correctly pointed out.

2. I don't seem to catch your rationale behind the two instructions you wrote. The first `getPose` lets you retrieve the current location of the end-effector in the Cartesian space (translation _x_ and orientation _o_), whereas the second `goToPose` allows you to drive the end-effector towards a different pose. In a typical tracking scenario, you aim to let the end-effector follow a predefined trajectory, hence you'd need to use only `goToPose`, feeding it with points sampled from the desired trajectory with the correct timing. Therefore, `getPose` will only give you the chance to monitor how well the end-effector is performing tracking. Never pass what `getPose` gives you on to `goToPose`: it doesn't make sense :smiley: 

  You might find useful reading also #30 on the same topic.

3. A value of `0.1` set for the `in-target` tolerance is extremely huge. It does roughly mean: "_let's stop  at around `10 cm` from my target_". Thus, I'd recommend you to use something similar to `0.01`, for instance, to begin with.

  Finally, the robot can get stuck and never reaches its target for too many reasons for me to attempt any guess: some joints might be in `hw-fault`; the stiffness and damping set to crazy values; the Cartesian controller not properly configured; so on and so forth...

  Are you using the robot or the simulator? It might help your investigation to significantly streamline the code you're using and just try out very simple snippets to accomplish only one point-to-point movement.

misaki43, 2016-08-11:
1. Ok thanks for this explanation. Then, when you call "goToPose", the computation of all the required "qd" are computed and then, the control is done with this same formula, right?

2. I think that what I want to do is not clear, so here is the explanation. The robot has no predefined trajectory to follow. The robot has to be totally compliant and it has to follow the partner's guidance on the arm limb. But the problem is that it is not totally compliant (not sufficiently to be able to move its arm easily). I assume it is because of the "virtual spring" that impose an equilibrium position (qd) that doesn't follow the « partner's movement ». 
Thus, I tried to cheat by modifying the qd position to the actual position (qd = qactual) in the hope that the partner can move its arm easily (totally compliant since qd near to q). But it doesn't work (when we don't move the robot, the arm robot move too much, as if there is noise). 
A cleaner way could be also to change the value in the 'setImpedance' function to put the arm in a more compliant mode, but as it is written in the tutorial, we cannot play randomly with the stiffness and damping values and I don't know how to choose these values to  have a totally compliant arm. 
I don't understand the relation with #30 even if it will be useful for me soon.

3. For the 10 cm, I know it was extremely huge : at the beginning I putted 0.01, but since the algorithm was blocked on the « waitMotionDone » function, I tried with 0.1 to verify if it really wasn't able to achieve the target. Thanks to this trial, it seems that the problem is not linked to the no-achievement of the movement. Indeed, the robot should have no problem to achieve a target that it has already achieve at the beginning (initial position) with 10 cm of allowed error, right?
The problem cannot be linked to hw-fault because I tried this code on the simulator.
I use the real robot, but also the simulator when I change the code to be sure to not damage the real one.

pattacini, 2016-08-11:
1. Yep
2. Then, you definitely need `zero-g` for your task, which is realized by means of pure `torque` control, as per the table above. I didn't really understand what your goal was in fact. Dump off **IM**, Cartesian control and take a look at force-control related demos available from within [**`icub-basic-demos`**](https://github.com/robotology/icub-basic-demos).
3. Would you mind posting on [**Gist**](https://gist.github.com/) the very minimal snippet of code causing the problem you have with the simulator? This is an example of [Gist snippet](https://gist.github.com/pattacini/6572864f8ec97c7f4639) you might draw inspiration from.

dogoepp, 2016-08-12:
I'm also curious about this topic. If I understand correctly Oriane, she would like that one can do some sort of teaching by demonstration by moving iCub's arm. My guess is that for such a use case one would want the arm's actuator to apply only the torque necessary to compensate for the gravity (and coriolis force ?).
I guess that we can compare it to the way humans would behave in such a case. Indeed, if someone takes my arm to show me a move, I won't fully relax it and let it fall. I'll spontaneously compensate for gravity and comply the forces applied by the other person on my arm.

How would you do that with the iCub ? Is there a service we can call to get the torques to apply in order to compensate for the gravity ?

misaki43, 2016-08-12:
Hi @pattacini 

2. Ok so in this force control demo uses what I need :
```
iimp->setImpedance(j,0.0,0.0);
imode->setTorqueMode(j);
```
In that case, the arm robot move as I want (is it what you call zero-g?).
I cannot dump of the Cartesian control as you suggest because I need it to record the trajectory (getPose). But It is not a problem since it doesn't perturb the torque control mode.
Moreover, when I will have to play back the trajectories, I will play with the soft/medium/hard spring parameters given by the demo, so no problem anymore.
So thanks, it is what I need !
This demo is really clean and understandable, so it could be interesting to add a link to these basic demos after the list of the icub-tutorials in this [page](http://wiki.icub.org/brain/icub_tutorials.html) , like a “going further” link.

3. About my last problem, the code line that provoke the problem is [here](https://gist.github.com/misaki43/23082a273455e677c536342593320075), or  if you want to launch it [here](https://gist.github.com/misaki43/5014f5076118ab134cc8c9ad89c3ea17) 

Thanks !

pattacini, 2016-08-12:
@dogoepp 
yep, exactly, you'd need a couple of basic components that lets iCub compensate for internal dynamics and gravity (aka `wholeBodyDynamics` and `gravityCompensator`). Examine our `icub-basic-demos` to find more on this.

@misaki43 
I didn't realize that @misaki43 was you, Oriane :smile: 
Glad to know you found the demos useful. I've just put the link in the tutorial page as you suggested: see https://github.com/robotology/icub-tutorials/commit/4d5d7bba87fb2d7a90d35c0b06746e4696e1918a. Wait until the doxy daemon gets triggered to see the updates online.

Also, you might consider looking at [**`ARE`**](http://wiki.icub.org/iCub/main/dox/html/group__actionsRenderingEngine.html) for kinesthetic teaching and action imitation. There are _ad-hoc_ functions for that.

Finally, I'll get back to you with my comments as soon as I'll have time to go through your snippets.

pattacini, 2016-08-12:
@misaki43 actually, it'd help if you put your name in the GitHub profile :wink:

pattacini, 2016-08-16:
@misaki43 

Find on **Gist** a few [comments](https://gist.github.com/misaki43/5014f5076118ab134cc8c9ad89c3ea17#gistcomment-1850631) of mine.

misaki43, 2016-08-22:
@pattacini 
Hello, I regret I have choose this name, sorry!
I will look at ARE soon, but I have already read your comments, and answered them, thanks!

pattacini, 2016-08-22:
Never mind @misaki43 
I've just added further comments to your gist.

I think we can close this issue by now, because we made somehow clear a bunch of control aspects you were inquiring. Though, feel free to open new tickets for any other matter, of course.

closed 2020-12-08 by pattacini