robotology/community #133: Unhandled exception running cartesian-interface code for icubsim

Anouk92, 2016-08-24:
When I try to run the castesian-interface code for the icub simulation that is provided on the Teaching-master github, I get the following exception: 

" Unhandled exception at 0x00D831CA (msvcr120d.dll) in cartesian-interface.exe: 0xC0000005: Access violation reading location 0xBA006574."

The program breaks in the code on the following line: "if (!client.open(option))" [= line 70]

The system I use is Windows 10
yarp version 2.3.66 v12 x86
iCub version 1.4.0 v12 x86
CMake verison 3.6.1 x86
Visual Studio 12 2013, also 32 bits

Can someone tell me what I need to do to get the code running?

Thanks!

pattacini, 2016-08-24:
Hi @Anouk92 and welcome on board :smile: 

I've just tried out that code on my `Windows 10` with `Visual Studio 2013` and `yarp/devel` + `iCub/master` compiled at `32 bit` from sources. No problems spotted. The only significant difference I see is then that you used `yarp` and `iCub` pre-compiled binaries.

The exception you got indicates that there's a **memory safety violation**, which is in turn likely due to a [_dangling pointer_](https://en.wikipedia.org/wiki/Dangling_pointer). Given that the [`open`](https://github.com/robotology/icub-main/blob/master/src/libraries/icubmod/cartesianController/ClientCartesianController.cpp#L90) method of the client doesn't allocate and/or reference explicitly any memory via pointers (as far as I see, at least), I would believe the failure might be caused by `yarp` calls themselves.

Therefore, did you already check the **integrity of your installation**? In other words, a possibility would be also to run a different, simpler [**tutorial**](https://github.com/robotology/icub-tutorials/tree/master/src/motorControlBasic), which does some basic operations with ports and motor interfaces.

Would you mind reporting on the results you'll get with this test?

Anouk92, 2016-08-25:
Thank you for your response!
I've tried several other tutorials, also the one you showed, and most of them seem to work. However, when trying the run the gaze-interface tutorial, I found that I could run the 'mover' project fine but the 'tracker' object gave the same exception: 

"Unhandled exception at 0x016731CA (msvcr120d.dll) in tracker.exe: 0xC0000005: Access violation reading location 0xBA006574."

Now the code breaks at line 58: 'if (!clientGaze.open(optGaze))', so again at the open method of the client.

pattacini, 2016-08-25:
I and @Anouk92 had a debug stint, finding that the run-time failure is caused by a mismatch between the user code compiled in `Debug` and the `yarp` + `iCub` binaries shipped in `Release` (most likely). In fact, the same tutorial compiled in `Release` runs just fine.

@mbrunettini @drdanz 
Are these considerations correct? Are the pre-compiled binaries for Windows actually distributed in `Release`? Is there any recommendation in the wiki saying that we need to compile the user code in `Release` mode while linking against those packages?

pattacini, 2016-08-25:
Issue moved to [robotology/icub-packaging #4](https://github.com/robotology/icub-packaging/issues/4) via [**ZenHub**](https://www.zenhub.com/)

closed 2020-12-15 by pattacini