robotology/community #134: Updated documentation for FW update of ETH robots

marcoaccame, 2016-08-25:
Hi all,
I have just updated the instructions for performing FW update on the ETH robots.

The new instructions can be found in the howto files inside folder https://github.com/robotology/icub-firmware-build/tree/master/ETH.

The files are now also linked by http://wiki.icub.org/wiki/Firmware#ETH_robots.

serena-ivaldi, 2016-09-29:
Dear all

we just updated the firmware following the instructions, but when we were updating the firmware with the CanLoader, we saw in the debug messages of the program on the terminal that there was a complaint about not having a recent version of the eUpdater module of the firmware.
We have been reading the instructions on the link you pasted (https://github.com/robotology/icub-firmware-build/blob/master/ETH/howto-eth-boards-programming.txt) but there is no mention of a procedure to update this eUpdater. Shall we do it?

@misaki43  @dogoepp  @paraschos

pattacini, 2016-10-02:
@marcoaccame could you give a look at this?

paraschos, 2016-10-07:
Something useful on this side: https://github.com/robotology/icub-firmware-build/blob/master/ETH/howto-eth-boards-fundamentals.txt

paraschos, 2016-10-07:
So if I get it correctly, to upload the eUpdaer we have to do the following:

- Boot the boards, click discover twice (i.e. load the eUpdater at ``START'')
- Flash the eApplPROGupdater as firmware (i.e. at ``DEF'')
- Jump to eApplication, a.k.a. ``DEF'' (or just reboot everything **Do not click twice at the discover**)
- Flash the new eUpdater (while the eApplPROGupdater is running)
- Reboot, click discover twice 
- Flash the proper firmware 
- Reboot and is it done

paraschos, 2016-10-07:
@serena-ivaldi I just checked 

 https://github.com/robotology/icub-firmware-build/blob/master/ETH/howto-eth-boards-programming.txt

Isn't it that Sec. (7) does the `eUpdater` update?

pattacini, 2016-10-12:
Hi @serena-ivaldi @paraschos @marcoaccame 

What's the status of this?

paraschos, 2016-10-12:
@pattacini Not sure, I am in Darmstadt. @serena-ivaldi Did you finish it?

@marcoaccame the new documentation is appreciated. Really clear! However, if you are in a hurry it is quite lengthy for checking it for the first time. Maybe you would like to have a quick n dirty version on the top of the howto with just the commands. 

Q: What is the order that of board chain? Is it the same as the IP (increasing from 1 to 11)? 
I just want to know to make debugging easier if ever needed... Maybe it should also be in the documentation.

marcoaccame, 2016-10-13:
Hi all,

sorry for the delay. 

In short.

1. Hi @serena-ivaldi, for now, there is no need to update to the new eUpdater (and eLoader), 
2. but if one wants to do it, the documentation can be found on section (7) of https://github.com/robotology/icub-firmware-build/blob/master/ETH/howto-eth-boards-programming.txt. And I am available for support during the update.
3. Hi @paraschos, the quick-n-not-so-dirty version of instructions is in https://github.com/robotology/icub-firmware-build/blob/master/ETH/howto-eth-boards-simple-fw-update.txt. I have decided to describe most common operations and to discard advanced ones.
4. The chain of ETH boards can change from robot to robot. The correct action would be to add the HW schematics in a file inside robots-configuration repository.  For your robot the addresses are: 
`10.0.1.104 - 10.0.1.1 - 10.0.1.2 - 10.0.1.3 - 10.0.1.4 - 10.0.1.5 - 10.0.1.6 - 10.0.1.10 - 10.0.1.7 - 10.0.1.8 - 10.0.1.11 - 10.0.1.9`

In more details.

3. The canLoader does not really complain. It just tells:  `[WARNING] remote board did not sent any ack to command 0x20. it may have a version of eUpdater older than ...`. The reason is that in the summer there has been a retro-compatible change of protocol between the canLoader and the eUpdater and it was a good thing to add some debug prints about what the canLoader finds on the boards on the various robots.

4. The change of protocol is only a piece of some activities aimed at a general improvement of the FW programming tools. Eventually (end of year), there will be a new GUI which manages both CAN and ETH boards, hopefully with better and clearer modes of operation. The new GUI is wip under the branch fwupdater2 of icub-main / icub-firmware-shared. 

5. The new GUI will behave in a slightly different manner, will have a simplified layout for common operations and a complete one for advanced users. There will also be a bash script to perform a simple complete FW update of the robot. Thus, completely new documentation is required. 

6. The new GUI will still manage old versions of eLoader / eUpdater even if a full FW update will give extra functionalities.

7. In order to offer enhanced features to this GUI, we needed to extend / review the protocol capabilities between the programs running on pc104 (aka: ethLoader, canLoader, the new GUI) and FW running on the ETH boards (eLoader, eUpdater, eApplication). 

8. The new protocol is already used by ethLoader and canLoader which also guarantee retro-compatibility with old versions of FW. Also, we have released new versions of eLoader (v 2.20), eUpdater (v 2.20 and now v 2.21) and application because with them the ethLoader can retrieve some more information from the ETH boards. But older FW versions still work as they did before.

9. Here is an example of what you can see in both cases: (a) keeping old FW, and (b) upgrading eLoader + eUpdater. Board 10.0.1.1 has an old FW whereas 10.0.1.6 has the latest FW. The new eLoader FW v 2.20 allows you to recognise the type of ETH board. The new eUpdater FW v 2.20 allows to retrieve more information about what is running on the board and to edit the field Info. At the moment it is not that much which I believe can justify an update. 
 
[image]

pattacini, 2016-10-18:
Thanks @marcoaccame 
Closing; feel free to reopen it again if needed.

closed 2020-12-12 by pattacini