robotology/community #136: Questions about Debian image 6.0 for pc104

gsaponaro, 2016-09-06:
Since we haven't changed the Debian image or the USB pen on iCubLisboa01 in ages (it's a CAN robot), and since lately we experienced some problems when shutting down the robot and the PC104 (which might or might not be related to the USB pen or OS, still investigating, I'll open another issue on that), we decided that it would be a good idea to install the new Debian image in any case.

We are following the official manual at http://wiki.icub.org/wiki/The_Linux_on_the_pc104 in order to prepare and configure the latest image (icub-live_6.0-16.04.05-rt-amd64.zip) from a Linux machine.

- the manual suggests using the FAT32 filesystem. Any particular reason for that? Can we use ext4 as we've been doing the last few years with the old image?
- we ran ```sudo bash makeboot.sh /dev/sdc1``` (sdc1 being the USB pen partition), it gave us [this output](https://gist.github.com/gsaponaro/10c9f5179a92776170841115e4198d10) which seems ok.
- we need to customize some files (```/etc/network/interfaces```, ```/etc/fstab```, ```/etc/hosts```) due to our cluster and network setup (our IPs are different from the ones used at IIT), but it is not clear to me how or when to customize those files in this image 6.0. When I boot the Debian live, I see this:

[image]

[image]

Up to now, we have been using the old image 4.0 "Open Call" with ```/etc``` files modified after burning the image on the USB pen.

/cc @mbrunettini @lorejam @vicentepedro

mbrunettini, 2016-09-06:
1) Fat32 - first the booting system we uses has been tested only on FAT32, then it's not the root filesystem: It is only the system were the system image resides. Why do you want to try a live system with EXT4?
2) it seems OK
3) the system is a live, but with persistency: this means that all the modification to the original filesystem will be stored in a persistency file and the not lost after a reboot

mbrunettini, 2016-09-06:
Going deeper into details:
1) the root filesystem is aufs
2) the persistency file is ext4
3) in theory, it can be possible to format the USB with other FS, but there will be no practical changes (the above will stay the same) apart making the system harder to boot (and running on a not tested configuration)

This replies to your questions?

gsaponaro, 2016-09-06:
Thanks for the informative replies. Regarding the filesystem type, I am now re-starting the whole procedure with FAT32, just to follow the manual 100%.

I understand the concept of live system with persistence, but something more basic is still escaping me. I am selecting the first boot option "Live (rt-amd64)" (in my first screenshot) and as a result I enter the basic system of my second screenshot, which doesn't have an ```/etc``` directory at all. How do I proceed from here? Ideally, I would like to see the network configuration files, so that I can customize them.

Just as an extra information: for now I am doing this while booting the pen on a desktop computer (which is neither the iCub laptop, nor the pc104 just because it's not very practical to connect a monitor there). After I manage to do the first basic customization (network address and DNS, etc.) I plan to put the pen into the pc104, and then proceed with further configuration there (```.bashrc_iCub```, etc.).

mbrunettini, 2016-09-06:
Please avoid to use the same live system on two different machines : the live system creates a new  network interface each new ethrnet hardware it founds (it's not our design, it' the debian live system). So you'll have eth0 and eth1 on the first machine (the workstation, in your case) and eth2 and eth3 on the secnd (the PC104). then networking won't work on PC104. (this can be solved by cleaning the file /etc/udev/rules.d/70-persistent-net.rules)

Then, normally you don't see the boot menu you sent me, so on the PC104 it's always the default (Live).

Finally, it's not possible the you don't have the /etc in the filesystem, the screenshot you sent is from the root home directory. Please try again.

gsaponaro, 2016-09-06:
OK for the network interfaces and the caveat to not use the live system on different machines. However, for now I would still like to try using & configuring the USB pen live system from the workstation first, and after solving my current problem cleaning ```/etc/udev/rules.d/70-persistent-net.rules``` as you suggest. (If that doesn't work, I'll ask our technician to plug a monitor to the PC104.)

Anyway, using the workstation I re-formatted the USB pen drive to FAT32 (it's a new SanDisk 32GB pen and I used the slow thorough formatting from the Ubuntu Disk Utility tool). I followed again the manual. **```makeboot.sh``` worked (I answered 'y' to all questions, including mbr and SYSLINUX, correct?)**, [here is the new makeboot.sh log](https://gist.github.com/gsaponaro/ea6fbeb4ff5e740ad4e9e2c16ff6c083).

Then I booted the live system on the workstation, and I observed the following notice about ```/etc/rc.local``` in the last line (probably not important):

[image]

After a couple of minutes the boot process completed, I entered the live system, and again I could not see ```/etc```:

[image]

You wrote:
_Finally, it's not possible the you don't have the /etc in the filesystem, the screenshot you sent is from the root home directory. Please try again._

I confirm that the screenshot is from the root directory of the live system (running on the workstation, which is the only machine that so far this live system has "known"). Any other suggestion? Thanks.

mbrunettini, 2016-09-06:
The following prompt:
icub@pc104:~$ la -la
States clearly that you are inside the home of the icub user (**~**)
And moreover the file .bashrc_iCub is places only in the iCub home.

As for the /etc/rc.local is normal if the pc104 is unable to contact icub-server (10.0.0.1)

gsaponaro, 2016-09-06:
D'oh! Basic mistake, sorry. :) OK, will try again and update.

gsaponaro, 2016-09-06:
Running the live system on the workstation, I have successfully customized the system to our needs (replacing 10.0.0.1 with our own server IP or hostname): ```/etc/network/interfaces```, ```/etc/hosts```, ```/etc/rciCub.d/mounts.list```, ```/etc/rciCub.d/S90_mount-remote-fs.sh```, ```/etc/resolv.conf```.

Just one networking-related question left to close the issue: in order to clean ```/etc/udev/rules.d/70-persistent-net.rules```, I have to comment or remove the line containing the "offending" ethernet address as shown below, correct? Assuming that I correctly configured all the files listed above, is there any other necessary step before I move the pen drive from the first machine (workstation) to the PC104?

[image]

At the moment our robot is busy, but as soon as possible I'll try on the PC104 (without a screen, hopefully it won't be necessary) and report back.

barbalberto, 2016-09-06:
after you log into the pc104 for the first time, you have to set the new address of the yarpserver with:

yarp conf <ip_address_of_the_server>  10000

gsaponaro, 2016-09-06:
Thanks for the prompt help, @mbrunettini. Looks like we'll have to connect a VGA cable to the PC104, which will take some time. Marking as solved for now, will reopen if necessary.

mbrunettini, 2016-09-07:
You can delete or comment the last two rows in 
<pre>
/etc/udev/rules.d/70-persistent-net.rules
</pre>
Good, let us know

gsaponaro, 2016-10-21:
Update: we made the PC104-VGA connection and we got our image to work.

Initially we had a few issues, originating from the fact that unfortunately we had to use the live system on two different machines (i.e., burn the image on one machine, then put it on PC104): 1) network interface number (eth0, eth1 etc.), fixed thanks to your previous clarifications; 2) changing the password was not permanent after reboots (persistence problem), but after fixing the previous point and customizing other files, this was solved too; 3) trying to make a fresh compilation of yarp master on the PC104 (after deleting the old cache) gave an error related to ```libdl.so``` not found. We installed ```libc6-dev-i386``` and all was good. Not sure if it's a good idea to add that package to icub-common for everyone in future versions?

Thanks again. :)

mbrunettini, 2016-10-21:
The libc6-dev-i386 is for 32 bit.
Can you please remember me which version of PC104 Image are you using?

gsaponaro, 2016-10-21:
1) iCub LIVE image for PC104 version 6.0-16.04.05

2) I noticed that the kernel reports the following errors, not sure if they are serious:
```
$ grep rror 161012_iCubLisboa01_pc104.txt 
[    0.307858] pci 0000:03:04.0: BAR 2: error updating (0xfdc00000 != 0xdc00000)
[    0.307969] pci 0000:03:04.0: BAR 3: error updating (0xfdc10000 != 0xdc10000)
[   14.631954] Error: Driver 'pcspkr' is already registered, aborting...
```
The full kernel log is here: https://gist.github.com/gsaponaro/47dfe7b1a9338d925a7d515f4564e256

3) Regarding libc6: now, having removed libc6-dev-i386, I cannot reproduce the problem anymore (because we did apt-get update & upgrade on the live system in the meantime). Anyway, this morning the image 6.0-16.04.05 was in a clean fresh state, and compiling yarp master complained about libdl.so not found. libc6-dev was already installed, so I tried with the 32-bit package version (admittedly not the best solution, I have removed it now).

mbrunettini, 2016-10-21:
Ok, then: I was really strange to me that a i386 package was missing on a 64 bit architecture... We tested the image and no packages were missing in order to compile YARP.
libdl.so ( used for dynamic linking)  is already installed.
Regarding the kernel, I cannot open the full log from the link you sent, but It is better if you see something else not working (the pcspkr already loaded does not seeems a problem), if you open a new issue. Keeping all in a big issue is misleading and creates confusion.

gsaponaro, 2016-10-21:
Sure. This issue is solved, we just tested some robot demos with the new image and all is well. If anything happens I'll open a new one, thanks again.

closed 2020-12-15 by pattacini