robotology/community #137: Differences between "my" URDF model and DH parameters in iKin

vicentepedro, 2016-09-09:
Hi Everyone!

I noticed a difference in the URDF model that I'm using and the kinematic chain defined in iKin. 
I inserted the right arm and torso joints at zero position (changing the limits in the iKin library) and there is an error around 0.8 cm and 15 degrees in the roto-translation matrix (torso to end-effector) .

I saw this [repo](https://github.com/robotology-playground/icub-model-generator#icub-model-generator) and I was wondering if I can use it to generate a new model of iCubLisboa01 from the DH parameters. (Arms V1.0 and Head V2.0)

Thanks in advance,
Pedro

traversaro, 2016-09-09:
Hi @vicentepedro ,  
   which iCub URDF model you are using? Where does it come from? 
  Thanks,  
             Silvio

vicentepedro, 2016-09-09:
Hi @traversaro 

I'm using a model made a couple of years ago here in Lisbon based on the CAD of the robot, I suppose.
You can find it [here](https://github.com/vislab-tecnico-lisboa/icub-moveit/tree/master/icub_description).

Thanks

pattacini, 2016-09-09:
Hi @vicentepedro 

First off, I would do a consistency check against our URDF models of iCub available in [**`icub-gazebo`**](https://github.com/robotology/icub-gazebo).

@traversaro please correct me if I'm wrong.

matejhof, 2016-09-09:
btw., @pattacini , (how) does `iKin` handle V1 / V2 / V2.5 robots?

pattacini, 2016-09-09:
@matejhof look at the limbs' constructions (e.g. [`iCubArm`](http://wiki.icub.org/iCub/main/dox/html/classiCub_1_1iKin_1_1iCubArm.html#ab6f2d2bd4a7ef453110a3555158d6799))

vicentepedro, 2016-09-09:
Hi @pattacini 
I will try to do the same test with a model in icub-gazebo repo, however, I don't know which model should be tested according to our iCub version - arms V.1 and Head V.2.

pattacini, 2016-09-09:
Good point @vicentepedro 
I'm not into the _Gazebo_ story and don't know if the versions you're inquiring are there. Perhaps @traversaro can shed a light on this.

Anyway, if the `arms v1.0` are not provided, we could think of creating the corresponding URDF models. It should be fairly simple to get the differences from `iKin` and use them to generate new files.

traversaro, 2016-09-09:
I am afraid that the answers to this may be lost in time, but you know by any chance on which CAD Assembly the URDF from Lisboa is based on? 
At the moment, that URDF is the only model of the hands that we have, and so I would be curious of understanding if it was based on the assembly with v1 arms or with v2 arms. 

Beside this, consider that URDF link frames needs to be different from the DH frames due to mismatch of the requirements of the two formats (DH requires the link frame origin to be placed on the axis of the **child joint**, while URDF requires the liink frame origin to be place on the axis of the **parent joint**) so a mismatch between the two link frame is inevitable. Furthermore in the CAD models there is no concept of link frames, so the URDF generated that you are using could have completely different convention for the link frames placement. 

For this reason we could use the models generated by `icub-models-generator` provide the DH frames as "additional frames", with the names documented here: http://wiki.icub.org/wiki/ICub_Model_naming_conventions#iKin_Frames . I added preliminary support for generating the iCubLisboa01 model (see https://github.com/robotology-playground/icub-model-generator/commit/256d824d1762cce988760c77cc709b5ebee03758, accidentally commited on `master`) but I did not commit the generated model because I am unsure about the version of your legs.

traversaro, 2016-09-09:
To answer @pattacini 's question: the `icub-gazebo` repository contains only `SDF` files, that is the format fully supported by Gazebo. 
I originally generated those file for an iCub with v1 arms and v2 head/legs, but since when I generated those file there has been multiple manual modification of the model (for example for adding eyes and hands, with the hands taken from the URDF model that @vicentepedro was using initially!) but it should still have the arms v1 kinematics. 
Eventually I plan to clean all the mess up and generate all the models automatically to have a [single source of truth](https://en.wikipedia.org/wiki/Single_source_of_truth), see https://github.com/robotology-playground/icub-model-generator/issues/25 and https://github.com/robotology-playground/icub-model-generator/issues/29 .

vicentepedro, 2016-09-09:
Hi @traversaro 

I'm not sure about the CAD Assembly used in the URDF model generation. I will try to confirm it with a former student here at Lisbon and let you know ;)
However, I would bet in the V.1 since it is the arm's version here....

"Our" legs are V1, also. ;)

Thanks in advance for the new model.

traversaro, 2016-09-09:
The newly generated `iCubLisboa01` has been committed to https://github.com/robotology-playground/icub-model-generator/commit/e3b35c515acd31a1b31119a0dd3f4dcad8a0a82a .

Disclaimer: the model does not include hands or eyes, and at this moment is generated from the `iDyn` models, so any inconsistency between `iDyn` and `iKin` models would be reflected on the URDF generated model. Let me know if this solve your inconsistencies. 

The disclaimer that you can read at the `icub-model-generator` repo also apply:  https://github.com/robotology-playground/icub-model-generator/blob/master/README.md .

traversaro, 2016-09-09:
For reference: the iKin Torso + Arm chain should match the transform between the `root_link` <---> `r_hand_dh_frame` in the URDF file. 
Note to self: this check could be also inserted in the tests that we run on every model generated by the `icub-model-generator` in https://github.com/robotology-playground/icub-model-generator/blob/master/tests/icub-model-test.cpp .

vicentepedro, 2016-09-10:
Thanks again @traversaro 
The roto-translation is now correct ;) I will close the issue.

Do you plan to include hands and eyes in the model in a near future?

traversaro, 2016-09-10:
Including hands and eyes is something that I would want to have in the generator, but unfortunately I don't think we are able to implement it in the near future. 
To keep track of any update, please subscribe to this issue https://github.com/robotology-playground/icub-model-generator/issues/29 .

closed 2020-12-15 by pattacini