robotology/community #139: Acquiring left and right images in a synchronized way

gsaponaro, 2016-09-13:
Hi @ale-git, is there any way to acquire the two iCub camera images (at 640x480@30fps, RGB) in a synchronized way?

This would be useful for stereo vision, e.g., for @nmoutinho's online stereo calibration system (ICDL-EpiRob 2014 paper "Markerless Online Stereo Calibration for a Humanoid Robot"), whose current implementation at the moment is bypassing YARP, reading directly from the cameras' vendor drivers.

From the camera [datasheet](https://www.ptgrey.com/support/downloads/10140):
_Multiple Dragonfly2 cameras networked on the same IEEE-1394 bus are automatically synchronized to each other. The maximum deviation between cameras is 125μs._

Would this require some modifications to [yarp::dev::DragonflyDeviceDriver2Rgb](http://wiki.icub.org/brain/classyarp_1_1dev_1_1DragonflyDeviceDriver2Rgb.html)?

/cc @lornat75 @plinioMoreno @ruipimentelfigueiredo

ale-git, 2016-09-13:
Hi, as said by the datasheet, the two images are already synchronized by default because the two cameras belong to the same Firewire bus. 
Cheers, Ale

gsaponaro, 2016-09-13:
OK. I wrote [this small program](https://github.com/gsaponaro/tools/tree/master/checkCameraSynch) which basically does the following:
```
    ImageOf<PixelRgb> *yarpLeftImg = leftImgPort.read(true);
    ImageOf<PixelRgb> *yarpRightImg = rightImgPort.read(true);

    Stamp stampLeft;
    Stamp stampRight;
    leftImgPort.getEnvelope(stampLeft);
    rightImgPort.getEnvelope(stampRight);

    if (yarpLeftImg==NULL || yarpRightImg==NULL)
        return true;

    yInfo("left %f, right %f => deviation between cameras %f",
          stampLeft.getTime(),
          stampRight.getTime(),
          stampLeft.getTime()-stampRight.getTime());
```

Then, we did the following tests:

1)
On the PC104 we launch two ```yarpdev``` Dragonfly2 drivers (Bayer carrier, 640x480@30fps, correctly configured). We don't launch any ```camCalib```, ```yarpview```, ```frameGrabberGui2``` to make sure that we don't overload the network.

Then we connect the images to the test program but for some reason we don't get any information about timestamps:
```
yarp connect /icub/cam/left /checkCameraSynch/left:i tcp+recv.bayer+method.vng
yarp connect /icub/cam/right /checkCameraSynch/right:i tcp+recv.bayer+method.vng
```
Result:
```
[INFO]left 0.000000, right 0.000000 => deviation between cameras 0.000000
```

Same thing with ```tcp+recv.bayer+method.nearest``` or ```udp+recv.bayer+method.nearest```.

However, using carriers other than Bayer:

2)
If we simply use UDP (I know, not a good idea for 640x480@30, just for testing) we obtain 1 - 2 msec delay:
```
[INFO]left 1473783359.958674, right 1473783359.956829 => deviation between cameras 0.001845
```

3)
With TCP we get 10 - 20 msec delay, which is quite worse but still acceptable for our requirement of 30fps => max 33 msec:
```
[INFO]left 1473783466.221515, right 1473783466.238958 => deviation between cameras -0.017443
```

4)
If we also run ```camCalib```, ```yarpview```, ```frameGrabberGui2``` (not on PC104 but on other computers in the same Gbit network) things slow down quite a bit:

UDP
```
[INFO]left 1473784107.170071, right 1473784107.186243 => deviation between cameras -0.016172
```

TCP
```
[INFO]left 1473784158.625859, right 1473784158.674626 => deviation between cameras -0.048767
```

Questions:

- in test 1 (Bayer carrier) any idea why ```stampLeft.getTime()``` prints 0.000000 instead of a Unixtime value, but all other tests print a valid time?
- [to everyone reading, not just for the Genoa guys :) ] can you please run my test program with your robot, just to make sure we don't have any other problem e.g. network setup?
- any other suggestions or existing examples regarding YARP client code that uses two cameras in synch?

Thanks a lot.

lschilli, 2016-09-13:
- I always heard that it is not really possible to grab the iCub cameras fully synchronized. But maybe I am wrong / the info is outdated.

Just two suggestions:
- Why not disabling the debayering transport for testing the timestamps but just display/check the bayer image directly. This excludes the debayering code as source of the error.
- I would trace down the timestamps to their origin (i.e. the line in the corresponding cpp file generating them. Otherwise it is never 100% clear what the true semantics of a certain timestamp is). Also what about the "stamp" option or potential other timestamp related options in the dragonfly2_config_*.ini files? Are there any differences between your dragonfly2 vs. dragonfly2raw configurations?

pattacini, 2016-09-13:
Hi @gsaponaro 

Just a couple of comments to add to the discussion.

I don't know why with `Bayer` there are no time stamps available in the message. Is it expected or a bug? Perhaps @ale-git can give an hint here.

Regarding `UDP` vs. `TCP`, ideally you should always go with `UDP` given the fact that cameras are streamed with a specified rate and if you lose a packet it's generally better to wait for the next instead of asking for a re-transmission as it happens with `TCP`. Of course, this holds in the assumption of a network with a low value of packet loss.

Sustaining two uncompressed streams of `640x480x24` @ `30fps`, which in turn are likely to be duplicated by other modules in the net, can be a tough job even with a 1 Gbps. `MCAST` protocol might come at hand to alleviate this burden.

@lschilli when two perfectly synchronized streams are sent over the network, then they're out and the network rules in this respect, thus there's a very minimal chance to preserve the synchronization untouched at the receiver side. If splitting hairs is the ultimate goal (usually this is not the case), I'd see the following possibilities to mitigate this problem:

1. Using `Quality of Service` to increase the priority of those channels @apaikan 
2. Sending in one image the aggregation of the two original left and right frames, that is something already done by [`dualCamCalib`](https://github.com/robotology/icub-main/tree/master/src/modules/dualCamCalib) running aboard the `PC104` @drdanz 
3. Writing a custom code at the receiver side to `recover the synchronization` (a sort of `PLL`) at the cost of an increased latency (we most likely have to drop some packets and/or wait for others).

lschilli, 2016-09-13:
@pattacini I was not implying that images will magically arrive synchronized over network. Anyway, thanks for highlighting that point.

About the bandwidth issue I always used one camcalib module per vision machine, so the network only needs to handle the Bayer images and not send debayered images over network. If CPU load due to debayering is an issue (good debayering can take some, and additional filtering such as sharpening/saturation can take even more), it can also be offloaded to the graphics card (requires cuda, see e.g. [camCalibGpu](https://github.com/lschilli/camCalibGpu)).

gsaponaro, 2016-09-13:
Thanks for all the valid points! For us it would be enough to have a reasonably low delay < 33 ms, at this stage we're not interested in the splitting hair part. :)

@lschilli:
_Also what about the "stamp" option or potential other timestamp related options in the dragonfly2_config_*.ini files? _Are there any differences between your dragonfly2 vs. dragonfly2raw configurations?_
We do use the ```stamp``` option for both subdevice cases. However, I just noticed that there is a ```use_network_time 1``` option in our ```dragonfly2_config_{left,right}_bayer_640_480.ini``` (taken from the xml/ini default templates), will investigate about that one.
BTW, a new student who just started here is precisely going to use your ```camCalibGpu``` + gaze work, thanks for making it available, we'll keep in touch.

@pattacini:
Unfortunately Multicast is not working properly here, but UDP is (https://github.com/robotology/icub-support/issues/4). Indeed, UDP makes more sense than TCP for streamed images for the reasons that you give. I just tested the various carriers for curiosity, wondering if the resulting delays are comparable with other iCub lab networks around.
About ```dualCamCalib```, it's cool but I think that @nmoutinho needs non-rectified images.

Will wait for @ale-git's comments in general, and on why we don't get timestamps with Bayer. I suspect it's a bug.

pattacini, 2016-09-14:
With `UDP` you've experienced a delay of ~ `16 ms` with `camCalib`, `yarpview`... running as well. I don't know whether this is satisfactory for you, given that you didn't mention the packet loss percentage.

By the way: `640*480*24 / (1/30) * 2 bps` ~ `440 Mbps`, that is two uncompressed flows already amount roughly to half of the available bandwidth. If you add up other flows - as entailed by the use of other modules -, then it's quite expected that things get worse.

Let's wait for feedback on `Bayer` to repeat the experiments with compressed streams of images.

pattacini, 2016-09-14:
@gsaponaro 
Actually the tests can be carried out with `Bayer` in the meantime using `yarpdatadumper`.

barbalberto, 2016-09-14:
The other option we were thinking about some time ago was to have a single device grabbing both images from firewire on the pc104 and sending them together in a single port. The same shall be done for the usbcamera, kinect-like devices etc...

After some evaluation this option presented some disadvantage, like:
- A single streaming of both cameras fused together will not be compatible with a lot of software in yarp/iCub, which expects to work on a single frame. (In my understanding, but I may be wrong).
- Having 3 streams (one port for each camera and a port for the both camera together) will preserve compatibility but will consume double of the bandwidth and CPU. Also probably will not be doable... or also the grabber shall be modified.
- IMHO, letting the user choose which configuration to use at startup will end in a robot with 'unpredictable' / non-standard set-up.
- To implement this behaviour all grabbing device shall be modified. 
- It will probably work only on firewire cameras as most of the usb camera will not have HW synch feature and will not benefit from this change
-  others ....

Although none of this is a _reason to not implement_ a new device, my opinion is that it doesn't worth the effort since it is not a general solution.

Considering that framerate is at most 30fps, then this means (in my understanding, but I may be wrong) that is the delay between 2 streams is under ~20ms, then the streams can be considered in synch since no new images have been generated yet. I believe this synchronization can be achieved on client side.

While designing the rgbd client I was thinking to implement a class to synchronize the streams on client side, using different algorithms depending on the sources of the streams.

If this synch class will work, then can be used to synch any kind of streams, also 2 rgb streams, and can be used anywhere since it works on ports.
I think this will be a more general solution, usable also on usb cameras which does not have an HW synch feature (or with any type of data since, ideally, only envelope info will be required).

What do you think about?

Note:
- As @pattacini was pointing, use `UDP` with streaming in general.
- If using bayer there is no envelope, I guess this is a bug and should be fixed

pattacini, 2016-09-14:
@barbalberto having this kind of resync daemon running on the client would be beneficial in different circumstances. 👍 

My only concern is that it would be mostly of help in conditions where the network gets somewhat overloaded, that is exactly when introducing further latency (due to the resync) might make things even worse.

barbalberto, 2016-09-14:
Yes, efficency will be a must, I haven't tackled the implementation yet.

The uses case I was thinking originally was not related too much with network overload, but when the streams comes from different sources, like:
- depth only sensor (like http://structure.io/uses) + rgb only sensor
- rgb camera and a software computed depth (disparity or other)

In those cases I thought some synch will be required, and cannot be done from the source side.

ale-git, 2016-09-14:
@gsaponaro by using **use_network_time 1** you're forcing the driver to use the system time in the envelope, while using **use_network_time 0** the timestamp is taken from the Firewire hardware clock (when use_network_time is omitted, the system time is used as in the first case). Usually you want to use the netowrk time because the Firewire camera stamp, though precise, gives no information able to match the images with other component state, for example the position of the head when the image was acquired. About the lack of timestmap in bayer carrier mode, I agree with @barbalberto about a bug in the carrier. I also noticed that there is a bug in the Firewire hardware timestamp reading and I'm going to inform Paul about it.

pattacini, 2016-09-19:
@gsaponaro to speed up things, would you be willing to test whether the quick modification @ale-git proposed to fix the typo (reported above in the reference) will actually sort out the problem?

gsaponaro, 2016-09-19:
Sure, will do that soon and report here.

ale-git, 2016-09-19:
Hi @gsaponaro the timestamp field is not a pointer, thus the code should be:

uint32_t v = ntohl((uint32_t)m_pFrame->timestamp);

gsaponaro, 2016-09-19:
Thanks. Now the device with hardware time runs, but still no timestamp with Bayer carrier.

lornat75, 2016-09-20:
Some info that may be useful. With @apaikan we have done tests on cameras synchronization. If I remember correctly these were the resuts:
- The two cameras do get synchronized
- The time at which they are acquired tend to drift with time. To prevent this drift the two framegrabbers should run on the same core, this can be set using process affinity. Another solution could be to acquire the two frames from the same process/thread as suggested by Alberto previously.

lornat75, 2016-09-20:
Concerning timestams, I think we ran into a similar problem with the mjpeg carrier (cc @randaz81).

gsaponaro, 2016-09-20:
Thanks for all the information, I am closing this issue. We conducted further tests with @plinioMoreno, summary:

- using the UDP carrier with two 640x480@30fps streams we get acceptable synchronization between cameras (3-10 msec difference between them), also when we add ```camCalib``` and we use the rectified images - good for our needs :)
- I opened a separate issue about the Bayer carrier not providing an envelope: https://github.com/robotology/yarp/issues/909
- I opened a separate issue about what we observed when turning the Firewire hardware clock option on: https://github.com/robotology/icub-main/issues/373

pattacini, 2016-09-20:
Thanks @gsaponaro

closed 2020-12-08 by pattacini